jvcleave/ofxOMXPlayer #43: Blank screen when loading a second movie

Maigre, 2015-01-22:
Hi !
Thanks for all the recent improvments :)
It pushed me to re-factor my ofxOMXPlayer based "HPlayer".
I have an old bug, wich might be an ofxOMX concern (or maybe the way i use it..):

In textured mode (only), i first play a movie, then i use my custom 'next' function
to play the next one in my playlist. Then, for some particular movies, 
the screen stay black while the sound, getFrame, etc.. are working properly.
From there no matter if i next, or stop/play again, the video will not come back.
So i guess something broke up in the Engine, but i don't see particular error.

I do have this problem with a particular file:
the Sintel 480p mp4 trailer
https://download.blender.org/durian/trailer/

There is no problem with the BBB, or the Sintel 720p mp4 trailer!
It happens after reading the Sintel 480p, no matter what is the next movie.
I guess something broke during the playback of this particular movie...
preventing the next movie to play properly. But i can't figure what.

I tried both implementation of my 'next' function:
1°: with loadMovie(file)
2°: with stop(); / settings.videoPath = file; / setup(settings);

My 'stop' function is doing:
- close() / framebuffer.begin(); / ofClear(0,0,0,0); / framebuffer.end();

The player is instancied with a pointer.

It's not a big issue since i think it's related to a bad encoding ,
but it could be interesting to understand what is breaking,
and maybe prevent it or fire an error log...
Because the strange part is that the guilty movie do display properly,
it just prevent next ones to show there video.

Thanks again for the good work :+1: 

Best Regards,

jvcleave, 2015-01-22:
I just ran the trailer with another one of my movies (1280x720) with the example-playlist and it seemed ok

You can jump to the next movie by pressing 'n'

Notice it sets a `doLoadNextMovie` variable which is not immediately done but is done later in `update`

Maigre, 2015-01-23:
Ok my bad,
it was related to my stack of shaders.
I used to have a 'passthru' default shaders before optional other shaders.
I did that because the others shaders was optional, and in previous version
if i didn't have at least one default shader, it wasn't working.

Now it seems to work ok without this default shader (i still use a frame
buffer
optionaly redrawed into those optional shaders).
And removing this default shader did fix the problem mentionned above
while playlisting thru media with different resolutions.
Actually the playback is better like this :)

And live optional shaders (blur/etc..) are still working well.

I also moved the 'next' logic into update (but this only change wasn't
solving the problem at first try)

Thanks!

Best Regrads,

closed 2015-01-23 by Maigre