jvcleave/ofxOMXPlayer #44: [Suggestion] Getting 1080p in textured mode on Raspberry Pi 2

PTS93, 2015-02-06:
So according to the RPi2 announcement on the RPi Blog video decoding is a hell lot faster now:
"NEON-enabled multicore video codecs can be over 20x faster"

http://www.raspberrypi.org/blog/#raspberry-pi-2-on-sale
I couldn't find more info on why this is the case. Maybe the VPU changed with the new CPU, the GPU is still the same.

I can confirm that ofxOMXPlayer works just fine on the Pi2 so far, loading videos seems to happen quite a bit faster but without any modifications 1080p is still lagging in textured mode.

pikilipita, 2015-02-06:
Did you compiled you app on Raspberry 1?
If so does it means openframeworks raspberry pi 1 apps work fineon the raspberry pi 2?

PTS93, 2015-02-06:
I compiled on the Raspberry Pi 2. See this issue: https://github.com/openframeworks/openFrameworks/issues/3620

Though the same app binary compiled under the Rpi1 worked just fine on the Rpi2.

trentbrooks, 2015-02-16:
Hey @PTS93, which branch of OF are you using with ofxOMXPlayer for rpi2? I'm getting a blank screen for all ofxOMXPlayer examples except for 'example-multiple-players' (non-texture mode). Even then I only get 1 of the videos displaying. I'm using rc1's 084 test rpi2 branch mentioned on the forum- http://forum.openframeworks.cc/t/raspberry-pi-2-install-failing/18411/11

PTS93, 2015-02-16:
I'm using the 0.8.4 release for ARMv7 with the patch mentioned in this issue:
https://github.com/openframeworks/openFrameworks/issues/3620

Simply add this file: https://github.com/openframeworks/openFrameworks/blob/728e966aa69f48b6f7696514b61c068ba18c116c/libs/openFrameworksCompiled/project/linuxarmv7l/config.linuxarmv7l.rpi2.mk
to the 0.8.4 ARMv7 release and compile with:
```
$ export MAKEFLAGS="-j4 -s PLATFORM_VARIANT=rpi2"
$ make
```

My ofxOMXPlayer version is at this commit https://github.com/jvcleave/ofxOMXPlayer/commit/f948fccc77b0cc2f522d4d6013cad08839f02641

Compiling failed for me with the latest version of ofxOMXPlayer.

trentbrooks, 2015-02-16:
Yeah I'm exporting with the same flags. Will try with with the older commit for ofxOMX - thanks.

jvcleave, 2015-02-16:
What is the compile error for ofxOMXPlayer?

trentbrooks, 2015-02-16:
Hey @jvcleave - there are no errors for my issue. The video is just not displaying. I still have the debug info, and the timestamp is updating as expected. I'm guessing it has something to do with the GL texture, as non texture mode is working for 1 video.

jvcleave, 2015-02-17:
@trentbrooks - yeah - I was asking about @PTS93 's compile error making him use https://github.com/jvcleave/ofxOMXPlayer/commit/f948fccc77b0cc2f522d4d6013cad08839f02641

@trentbrooks - have you tried bumping up the GPU memory allocation? That is often the case for blank textures

trentbrooks, 2015-02-17:
Ah sorry - thought you were asking me. Yeah I tried bumping the memory split up to 256 - still blank though.

jvcleave, 2015-02-17:
@trentbrooks any luck with 384? (that's what I currently use)

trentbrooks, 2015-02-17:
Yes- thanks @jvcleave works perfectly now! Still a noob with the rpi stuff.

jvcleave, 2015-02-17:
@trentbrooks glad it works - that is a pretty high though (I only had it that high for multiple players)

 are you running X11?

trentbrooks, 2015-02-17:
Yeah I was actually - i should get out of the desktop and try.

PTS93, 2015-02-17:
I can try it out later again and give you details. I didnt want to open a bug just now as the whole thing is pretty boggy.

jvcleave, 2015-02-21:
Just got an RPI2 and everything seems to work fine

Using the latest Raspbian/OF 0.84 arm7 with the config.linuxarmv7l.rpi2.mk patch

I did update the RPI2 compiler to use g++ 4.8 (4.6 is the default for some reason) as some of my ofxRPiCameraVideoGrabber apps were getting compile errors (-fpermissive)

PTS93, 2015-05-25:
I had success playing back 1080p video without stuttering by overclocking the GPU and h264 decoder(VPU). See my forum post here: http://forum.openframeworks.cc/t/ofomxplayer-fading-full-hd-videos/19696/2?u=prof_sparkles