jvcleave/ofxOMXPlayer #45: ofxOMXPlayer.h: no such File

FlowBo, 2015-02-16:
i'm trying to compile one example but i always get the 
ofxOMXPlayer.h: no such File 
error
I could not find anything on the web so i ask you guy's what to do.
Thanks

jvcleave, 2015-02-16:
Which example? Did you put ofxOMXPlayer in your openFrameworks addons folder?

FlowBo, 2015-02-17:
yes i cloned the git to openframeworks/addons.
I tried 3 different examples:

openframeworks/addons/ofxomxplayer/example-shader
openframeworks/addons/ofxomxplayer/example-multiple-players
openframeworks/addons/ofxomxplayer/example-displayrect

always the same error.
i will attach an image of the exact error.
[image]

jvcleave, 2015-02-17:
rename the folder to `ofxOMXPlayer` 

Linux is case-sensitive

FlowBo, 2015-02-17:
Thanks! Now it's working!

closed 2015-02-17 by FlowBo