jvcleave/ofxOMXPlayer #47: getPixels() returns RGBA data

pikilipita, 2015-02-21:
I noticed this addon is using GL_RGBA format for the videos textures.
is there a reason not for using RGB format? Or is there a way to receive RGB formatted data when calling ofxOMXPlayer::getPixels() ?

jvcleave, 2015-02-21:
It really comes down to creating the EGLImage that the egl_render will accept - I couldn't get anything else but GL_RGBA to work. From what I have read using GL_RGB won't actually save any memory if that is what you are looking for,

pikilipita, 2015-02-21:
That's because the getPixels() function of the 'ofVideoPlayer' class returns an RGB format and I'm trying to keep my pixels manipulations code cross platform, in my case the raspberry and osx.
And yes, also because I tought it would save some RAM using RGB on the rpi.

jvcleave, 2015-02-22:
yeah - search for GL_RGBA on this page - OpenGL seems to converted everything to it internally anyway

https://www.khronos.org/opengles/sdk/1.1/docs/man/glTexImage2D.xml

closed 2015-02-22 by jvcleave