jvcleave/ofxOMXPlayer #49: expected performance inside an FBO

eloimaduell, 2015-04-07:
Hi ! 

I'm trying to achieve a smooth and fluid video reproduction in a RaspberryPi2 .
My testApp is drawing the omxPlayer into an FBO, then drawing the FBO to the screen with a ofxHomography matrix multiplication.

If I create the FBO of 1280x720 i get a framerate of 59-60fps.
But if i create the FBO with ofGetWidth() and ofGetHeight() the app runs at 30-35 fps, and the video looks less smoothly playing then with 1280x720. 

I'm using OF084 libs, Raspbian 3.18.7-v7+ , OMXPlayer  337004e

This is my memory partition :
  $ vcgencmd get_mem arm
  arm=624M
  $ vcgencmd get_mem gpu
  gpu=384M

I do use export MAKEFLAGS="-j4 -s PLATFORM_VARIANT=rpi2".
I'm not playing from X, but from command line .

I don't understand why i'm getting this  : 
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1680x1050 
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1680x1050 
When I'm using a FullHD monitor (1920x1080) ?
?¿ Seems like it's not detecting the display as a 1920x1080?)

MediaInfo log for my file . The file was rendered with Adobe After Effects.

General
Complete name : /Users/eloimaduell/Desktop/MIES_2015/RASPBERRY_MIES/VIDEO_CODECS/test2.mov
Format : MPEG-4
Format profile : QuickTime
Codec ID : qt
File size : 91.3 MiB
Duration : 1mn 41s
Overall bit rate mode : Variable
Overall bit rate : 7 549 Kbps
Encoded date : UTC 2015-04-06 20:58:45
Tagged date : UTC 2015-04-06 21:01:21
Writing library : Apple QuickTime
©TIM : 00:00:00:00
©TSC : 25
©TSZ : 1

Video
ID : 1
Format : AVC
Format/Info : Advanced Video Codec
Format profile : Main@L3.1
Format settings, CABAC : No
Format settings, ReFrames : 2 frames
Format settings, GOP : N=1
Codec ID : avc1
Codec ID/Info : Advanced Video Coding
Duration : 1mn 41s
Bit rate mode : Variable
Bit rate : 6 007 Kbps
Width : 1 280 pixels
Height : 720 pixels
Display aspect ratio : 16:9
Frame rate mode : Constant
Frame rate : 25.000 fps
Color space : YUV
Chroma subsampling : 4:2:0
Bit depth : 8 bits
Scan type : Progressive
Bits/(Pixel*Frame) : 0.261
Stream size : 72.7 MiB (80%)
Language : English
Encoded date : UTC 2015-04-06 20:58:45
Tagged date : UTC 2015-04-06 21:01:21
Color primaries : BT.709
Transfer characteristics : BT.709
Matrix coefficients : BT.709

Audio
ID : 2
Format : PCM
Format settings, Endianness : Little
Format settings, Sign : Signed
Codec ID : sowt
Duration : 1mn 41s
Source duration : 1mn 41s
Bit rate mode : Constant
Bit rate : 1 536 Kbps
Channel(s) : 2 channels
Channel positions : Front: L R
Sampling rate : 48.0 KHz
Bit depth : 16 bits
Stream size : 18.6 MiB (20%)
Source stream size : 18.6 MiB (20%)
Language : English
Encoded date : UTC 2015-04-06 20:58:45
Tagged date : UTC 2015-04-06 21:01:21

Other
ID : 3
Type : Time code
Format : QuickTime TC
Duration : 1mn 41s
Time code of first frame : 00:00:00:00
Time code settings : Striped
Language : English
Encoded date : UTC 2015-04-06 21:01:21
Tagged date : UTC 2015-04-06 21:01:21

It's the expected performance loss (from 60 fps to 35 fps) to use a bigger FBO (from 720p to 1080p?)

I've also experienced that drawing a smaller video makes framerate higher ? 

I'm not sure if i'm not using a proper sample file or if I've to accept that using a FBO to draw the ofxOMXPlayer might slow down things to this end (35 fps) ?

Related questions ... 
· Video Compression ? Better .MOV or .MP4 ?
· Video Compression ? Better 1 Keyframe per frame for a smooth fluid play ?

PTS93, 2015-04-07:
Like it already says in the readme, you can't playback 1080p fluid in textured mode. If you want 1080p playback you will have to use non-texture mode, which also means no access or control over the pixels drawn to the framebuffer, so if you need to use an FBO you will have to reside to textured mode. Anything above 720p in textured mode will potentially stutter.
So yes, the behavior is expected.

eloimaduell, 2015-04-07:
But the 1080p limitation in texture mode is understood as the resolution of the video or the size you want to draw it ? I thouhgt that it was related to the size of the video file. I have a 720p video but i'm trying to draw it into a 1080 display ....

And it's strange because without using the FBO i get a correct 60 fps, even i'm on texture mode . ?¿ So ofxOMXPlayer "alone" without any FBO involved run good and smooth.

eloimaduell, 2015-04-07:
I tried with the example_shader, where the OmxPlayer is drawn into an fbo to apply a shader pass. If i set in main.cpp to go to FULLSCREEN and 1920x1080 i get the same loss of speed, 35 fps. At 1280x720, 60 fps.

jvcleave, 2015-04-07:
I primarily run on a cheap HDMI TV at 720 - I do however force it into 720p mode

If you are getting unexpected resolutions (1680x1050) you may want to mess with /boot/config.txt - below is mine for the RPI1 and more info here: http://elinux.org/RPiconfig

```
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
#720p
hdmi_mode=4
#1080p
#hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=900

# for more options see http://elinux.org/RPi_config.txt
start_x=1
gpu_mem=384
core_freq=250
sdram_freq=450
over_voltage=2

#dispmanx_offline=1
```

eloimaduell, 2015-04-07:
Hi @jvcleave  ! 

Thanks for the config file ! Make thing very clear ;)

I still have the doubt if the limitation on texture mode between 1080 or 720 is related to the video file resolution of to the display resolution ?

I'll keep on trying to achieve a smooth playback ... lots of details to solve ....

By the way do you regulary update RPi core or firmware ? Do my versions look outdated ?
Is it necessary to update as stated here regularly : http://stevenhickson.blogspot.com.es/2013/02/updating-your-raspberry-pi-packages.html ?

Thanks for the info ...

eloimaduell, 2015-04-08:
Using @jvcleave config.txt and rebooting i got a 1280x720 display. On it, my app with the FBO runs smoothly at 60 fps ! So seems like drawing a 1280x720 video on a 1920x1080 monitor makes everything slow and not smooth playback even that OF says it runs at 30 fps.

jvcleave, 2015-04-08:
Unlike most I don't update unless I know it it fixes a specific bug for I am using a new feature. 

Good to hear it works - going to close as the issue doesn't seem to be with the addon

closed 2015-04-08 by jvcleave

nenadlazovic, 2015-09-15:
jvcleave,
yes issue is closed. my problem was /boot/config.txt. now 95% of videos play ok. there are still few that cause screen to go blank, but I treat these as exceptional cases.

Thanks.