jvcleave/ofxOMXPlayer #50: Segfault when reading mpeg2

cmarty, 2015-04-20:
Hi
I have mpeg2 video file. My rpi doesn't have unlocked mpeg2 decoder. When I try to play this movie, ofxOMXplayer segfault. Is there a way how to nicely not process video files which doesn't have appropriate video encoding ?
The link to the mpeg2 video is here:
https://www.dropbox.com/s/zdwlxy443neh7zd/mpeg2_segfault.mpg?dl=0

jvcleave, 2015-04-20:
I would try to covert to h264 with http://www.squared5.com/

Handbrake is also good for this

cmarty, 2015-04-20:
In my app the users can user their own video files so I don't have any control over what video files they want to play. So it would be perfect if I have a way how to stop processing file in some stage of decoding and display some message that video file is not supported instead of segfault.

PTS93, 2015-04-20:
You will have to determine the codec before loading the file. You can do this with a tool part of ffmpeg called ffprobe (or avprobe in libav). Run the file through this tool and check if the codec is h264, then depending on the outcome show your message or play the file.
Though you should also check if the file is also in a certain profile. Not all h264 profiles and levels are supported by the OpenMAX api on the pi.

jvcleave, 2015-04-20:
Ok, `ofxOMXPlayer:: setup` does return a bool so you could use that to check whether the file was able to be played.

I however have no real idea of where MPEG2 is crashing the player so in the meantime you may want to do as @PTS93 is suggesting and probe the file yourself before passing it to the player.

cmarty, 2015-04-21:
Thanks for tips. Meanwhile I have to do that way as PTS93 suggested.
But I think the best way would be to test ofxOMXPlayer:: setup in my app and in OMXReader:: Open do some checks to test if the file is in appropriate format. 
It would be also perfect if ofxOMXPlayer::loadMovie(string videoPath) is returning bool instead of void and return the setup() return code;