jvcleave/ofxOMXPlayer #51: change which video is playing with loadMovie(string filename) ?

eloimaduell, 2015-04-22:
hi ofxOMXPlayer !

I've been succesffully using ofxOMXPlayer but i'm having trouble to change the video which is playing (without stopping the app). 

I tried to make (once an OSC message with a message that tells which video to load ) : 

   /// LOAD MOVIE
    else if(address.find("load")!=-1)
    {
        
        string filename = m->getArgAsString(0);
        int fadeTime = m->getArgAsInt32(1);
        ofVec2f myPos = screenPosition;
        ofVec2f mySize = screenSize;
        
        cout << "pmOmxPlayer :: OSC :: load : " << filename << " : fadeTime : " << fadeTime << endl;
        
        
        omxPlayer->setPaused(true);
        omxPlayer->close();

        omxPlayer->loadMovie(filename);
     }

but the video doesn't get updated ? The app is not crashing, but the new video is not playing ?
I've looked at the "playListApp" example, but seems that is doing basically the same ... so LoadMovie() when needed ? 

Am I missing something ? 
(i've checked that all the videos that i'm trying to play have same resolution and they all play nicely in the ofxOMXPlayer )

Thanks

eloimaduell, 2015-04-22:
I've also tried without pausing and closing the player and same results ... I get a frozen frame from the video I load when I started the app ... ?¿

eloimaduell, 2015-04-22:
Ok I got it ! 

I was passing : "./videos/sample.mp4" to omxPlayer->loadMovie() and seems like he needs to receive the full path of the video file ! 

Is there a way that we can get the absolute path of the actual data folder so I can just send the present filename without the whole path on the OSC ?

e*

jvcleave, 2015-04-22:
````
string fullPath = ofToDataPath("", true);
````
Should do it

closed 2015-04-22 by jvcleave