jvcleave/ofxOMXPlayer #54: Implementing Step Frame Backward

rdob, 2015-05-24:
Hi jvcleave,
I want to ask if it is possible to add a stepFrameBackward function.
I'd like to use the ofxOMXPlayer as an perfomant and low ressouce image sequence player, therefor i have created an h.264 video with one picture by frame, and like to move it forward and backward using an rotary encoder, to get a stop motion effect. 

I have done investigations already, but it seems that the OMX library does not have such a function.
It would be great if you can help me to find a solution.

jvcleave, 2015-05-25:
I think you would have better luck with a set of preloaded image files. The decoder is really optimized for a streaming format so seeking/frame accuracy is pretty limited.

You main limitation would be the available GPU memory which should be ok if your sequences are fairly short.

rdob, 2015-05-25:
I've done already intense tests, the problem with the preloaded images on rpi ist the memory limitation and framebuffer speed, with openmax support its works pretty nice, forward frame stepping is perfect, is there really no way for stepping one frame back?

jvcleave, 2015-05-25:
Here is the reference for what ultimately controls stepping - I don't see a way to send it a negative or backwards value

http://www.jvcref.com/files/PI/documentation/ilcomponents/prop.html#OMX_IndexConfigSingleStep

yeah - I was thinking if the sequence was around 20 you would be ok

PTS93, 2015-05-25:
OpenMAX also support jpeg decoding on the VPU. An example for that can be found here: https://github.com/raspberrypi/firmware/tree/master/opt/vc/src/hello_pi/hello_jpeg

Technically you wouldn't need to load all images at once into RAM, only as many to play video while loading the upcoming frames, if you want a stop motion effect you probably don't want fluid 30fps anyway but less than that? If you decode the pictures on the GPU that should go pretty quickly, not as fast as h264 video but maybe fast enough. Maybe even ofxTurboJpeg works already, its pretty fast and can use NEON on the PI2.

Also have a look at the Raspberry Pi 2 in general, with 1GB RAM you have quite a bit available.
I agree with jvcleave though, going back frame by frame doesn't seem to be implemented in the OMX api and would also be very difficult without h264 encoded with an intra profile.

undessens, 2016-07-27:
ofxOMXplayerEngine.h seems to have a "rewind" function, multiplying the speed by -1.
Unfortunately, this function doesn't have an effect.  I am looking for the same function rdob actually