jvcleave/ofxOMXPlayer #55: Inverted Video playback when in textured mode

brendan87, 2015-06-02:
I'm running the examples from https://github.com/openFrameworks-arm/RaspberryPiGuideCAN and compiling them on for the Pi 2 before running. However, the omxPlayerTextured displays flipped videos. The non-textured example works fine though.

brendan87, 2015-06-02:
Scratch that - this doesn't appear to be an issue on the latest repository (probably should have tried that before posting this). Still - for future noobs it might not hurt to update the link on the site? Aside from that, thanks for writing this addon for the pi :)

closed 2015-06-02 by brendan87

nenadlazovic, 2015-09-06:
just got code from https://github.com/openFrameworks-arm/RaspberryPiGuideCAN dated nov 13, 2013, and video is inverted ...
what is the repo URL you used ?
thanks

PTS93, 2015-09-06:
he probably updated the linked commit of ofxOMXPlayer in the creativeapplication git. Just pull the latest code from this repo and use it to replace this: https://github.com/openFrameworks-arm/RaspberryPiGuideCAN/tree/master/ofxAddons/ofxOMXPlayer