jvcleave/ofxOMXPlayer #57: play stdin h264 stream

natxopedreira, 2015-08-01:
Hi

Im looking for a way to render a stdin raw h264 stream in openframeworks and Rpi. Its possible to use oxOMXPlayer to play this kind o stream.

Im using this to render the stream but i need to move to oF
https://bitbucket.org/befi/hello_video/src/c9c244be22cf447d76fc02820a51fb48a4e45bbb/video.c?at=default

data_len = read(STDIN_FILENO, dest, buf->nAllocLen-data_len);

instead of

data_len += fread(dest, 1, buf->nAllocLen-data_len, in);

jvcleave, 2015-08-01:
yes you can use .h264 files (like test.h264 that is on the RPi)

closed 2015-08-01 by jvcleave

natxopedreira, 2015-08-01:
Thanks for answer

Sorry i don't explain myself right.

I don't want to play a .h264 file, i want to play a raw h624 stream not a file.

Im using some like a digital video link to stream a rpicamera with 2,4 but not using tcp or udp, instead this just injecting the raw video data in the 2,4 signal (ref. https://befinitiv.wordpress.com/wifibroadcast-analog-like-transmission-of-live-video-data/#comment-632), this system uses a modified "hello_video" example on the rpi to view the stream. 

https://bitbucket.org/befi/hello_video/src/c9c244be22cf447d76fc02820a51fb48a4e45bbb/video.c?at=default

Just changing this

data_len = read(STDIN_FILENO, dest, buf->nAllocLen-data_len);
instead of
data_len += fread(dest, 1, buf->nAllocLen-data_len, in);

I wonder if its possible to do the same in your addon because i wanted to use this stream in a of app

Thanks!!!!!

jvcleave, 2015-08-01:
I think it would take some pretty extensive changes to [OMXReader](https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/OMXReader.cpp) in order for it to work the way you want

If you have no need for textures or the other features of ofxOMXPlayer I would probably just incorporate video.c into your app

natxopedreira, 2015-08-01:
Thanks!

In fact i just want to render the stream in oculus, and overlay some hud, seems like i need to find another way...