jvcleave/ofxOMXPlayer #58: when playing certain videos in "example-multiple-players", monitor shuts off

nenadlazovic, 2015-09-10:
When playing default Timecoded_Big_bunny_1.mov / Timecoded_Big_bunny_2.mov videos, example-multiple-players plays it ok.
When some custom videos are provided (instead of default bunny videos), monitor shuts off (black screen) and recovers once example-multiple-players process is killed. Unable to determine pattern. sometimes it play 2 videos, sometimes it does not.

is this a known issue ? are there any settings i need to try ?

thanks.

console output shows

root@raspberrypi:/home/pi/of_v0.8.4_linuxarmv7l_release/addons/ofxOMXPlayer/example-multiple-players# ./bin/example-multiple-players

[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1680x1050
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set
[notice ] ofAppEGLWindow: createSurface(): current renderer type: GL
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_
image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): mouse_fd= 5 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.2:1.0-event-mouse
[notice ] ofAppEGLWindow: setupMouse(): mouse device name = Logitech USB Receiver
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 6 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.3:1.0-event-kb
d
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name = CHICONY HP Basic USB Keyboard
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] ofAppEGLWindow: setupOpenGL(): peripheral setup complete
[verbose[] verbose] ofAppRunner: ofThreadGL ready
: - name: Thread 1 - Started Thread.
[verbose] ofAppRunner: Vendor:   Broadcom
[verbose] ofAppRunner: Renderer: VideoCore IV HW
[verbose] ofAppRunner: Version:  OpenGL ES-CM 1.1
[verbose] ofAppRunner: GLSL:
[notice ] ofAppEGLWindow: runAppViaInfiniteLoop(): entering infinite loop
[verbose]       banner1.mp4
[verbose]       jack-daniels.mp4
[verbose] listed 2 files in "/home/pi/of_v0.8.4_linuxarmv7l_release/addons/ofxOMXPlayer/video/"
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 4 MS
[verbose] OMXVideo::bool COMXVideo::Open(COMXStreamInfo&, OMXClock*, float, bool, bool) - decoder_component(0x0x1b5aed0), input_port
(0x82), output_port(0x83) deinterlace 0 hdmiclocksync 1

[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 29.9966
[notice ] bool OMXClock::OMXStart(double, bool): at pts: 0
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[verbose] OMXVideo::Decode VDec : setStartTime 0.000000

[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 7 MS
[verbose] OMXVideo::bool COMXVideo::Open(COMXStreamInfo&, OMXClock*, float, bool, bool) - decoder_component(0x0x2028518), input_port
(0x82), output_port(0x83) deinterlace 0 hdmiclocksync 1

[verbose] channels: 2 sampleSize: 2 sampleRate: 48000
[verbose] CPCMRemap: I channel map: FL,FR

[verbose] COMXAudio::Initialize Ouput bps 16 samplerate 48000 channels 2 device hdmi buffer size 384000 bytes per second 192000 pass
through 0 hwdecode 0
[verbose] PASSTHROUGH DISABLED, Audio codec aac channels 2 samplerate 48000 bitspersample 16

[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 30.03
[notice ] bool OMXClock::OMXStart(double, bool): at pts: 0
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[notice ] ofAppEGLWindow: runAppViaInfiniteLoop(): setting up notifications complete
[verbose] OMXVideo::Decode VDec : setStartTime 0.000000

[verbose] COMXCoreComponent::WaitForCommand OMX.broadcom.clock
                                  wait timeout event.eEvent 0x00000000
                                  event.command 0x00000000
                                  event.nData2 1

[notice ] ofAppEGLWindow: destroyNativeKeyboard()
[notice ] ofAppEGLWindow: destroySurface(): destroying EGL surface
root@raspberrypi:/home/pi/of_v0.8.4_linuxarmv7l_release/addons/ofxOMXPlayer/example-multiple-players#

jvcleave, 2015-09-10:
the `"Cannot display this video mode"` message is makes me think that your ofApp window size has an issue with the monitor

I am assuming you are using non-texture mode? Have you tried increasing the GPU memory?

nenadlazovic, 2015-09-10:
non-textured mode. GPU_MEM=192. 
In real app I am developing, i used fbset to get default monitor settings and passed these to ofSetupOpenGL().

Maybe, I should focus on monitor settings settings. I used strategy as defined in http://blogs.wcode.org/2013/09/howto-boot-your-raspberry-pi-into-a-fullscreen-browser-kiosk/
"set the internal framebuffer to as large as it can be, then detect the monitor’s capabilities and adjust."
Maybe that is the cause. 

Do you have suggestions of how to dynamically configure monitor resolution (if needed at all) ?

BTW: thanks for excellent addon.

jvcleave, 2015-09-10:
I would first try bumping up the GPU memory to even 384

I haven't had to mess with dynamically configuring it

nenadlazovic, 2015-09-10:
ok. .will try. 
do you recommend chanding /boot/config.xml, or just using default NOOBS  
      # NOOBS Auto-generated Settings:
      hdmi_force_hotplug=1
      config_hdmi_boost=4
      overscan_left=24
      overscan_right=24
      overscan_top=16
      overscan_bottom=16
      disable_overscan=0

jvcleave, 2015-09-10:
Here is what I use

https://github.com/jvcleave/ofxOMXPlayer/issues/49#issuecomment-90308103

I haven't used NOOBS yet - I typically use the Raspbian image

PTS93, 2015-09-10:
You can alter the resolution after boot with the `tvservice` application.
Just run `tvservice -h` for some guidance.

closed 2015-09-20 by nenadlazovic