jvcleave/ofxOMXPlayer #59: running openFrameworks / ofxOMXPlayer in portrait mode on Raspberry PI2

nenadlazovic, 2015-09-15:
As part of project, I need to run openFrameworks / ofxOMXPlayer in portrait mode on Raspberry PI2.
I tried few things
* add line display_rotate=3 to /boot/config.txt 
** framerate drops significantly (e.g. from 60fps to 5fps)
** console window is keep interfering with openFrameworks window (where ofxOMXPlayer is playing in)  
* openFrameworks ofOrientationToDegrees(OF_ORIENTATION_90_RIGHT)
** ofxOMXPlayer still plays in landscape mode

Any suggestion on how to have ofxOMXPlayer plays in portrait mode ?

PTS93, 2015-09-15:
I would encode the videos in 9:16 and try that.
Also try textured mode, in non textured mode the hardware encoder just draws how ever it wants to the screen, in non-textured mode a 9:16 video is the only thing that could maybe work.

nenadlazovic, 2015-09-15:
idea makes sense.

Concern is frame rate achieved in textured mode. We did play with ofxOMXPlayer in textured mode, we noticed that frame rate achieved drops significantly ... not 20,30% ... it drops 5-10 times (e.g. from 50 fps to 5 fps). 
Is that big of the frame rate drop expected in textured mode ?

jvcleave, 2015-09-15:
there is a `OMX_IndexConfigCommonRotate` that may be able to rotate it - I usually start searching [here] (http://www.jvcref.com/files/PI/documentation/ilcomponents/video_render.html)

Yes - above 720p the textured mode is much slower

jvcleave, 2015-09-15:
Change this method 
https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/OMXVideo.cpp#L523

to this

```
	OMX_CONFIG_DISPLAYREGIONTYPE configDisplay;
	OMX_INIT_STRUCTURE(configDisplay);
	configDisplay.nPortIndex = m_omx_render.GetInputPort();

    configDisplay.set = (OMX_DISPLAYSETTYPE)(OMX_DISPLAY_SET_DEST_RECT|OMX_DISPLAY_SET_SRC_RECT|OMX_DISPLAY_SET_FULLSCREEN|OMX_DISPLAY_SET_NOASPECT | OMX_DISPLAY_SET_TRANSFORM);
	configDisplay.dest_rect.x_offset  = 0;
	configDisplay.dest_rect.y_offset  = 0;
	configDisplay.dest_rect.width     = 1280;
	configDisplay.dest_rect.height    = 720;
	configDisplay.fullscreen = OMX_FALSE;
    configDisplay.noaspect   = OMX_TRUE;	
	configDisplay.transform  = OMX_DISPLAY_ROT90;
	m_omx_render.SetConfig(OMX_IndexConfigDisplayRegion, &configDisplay);
```

Make sure `configureDisplay();`  by editing the code here
https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/OMXVideo.cpp#L349

refer to this for other options
https://github.com/popcornmix/omxplayer/blob/master/OMXVideo.cpp#L635

nenadlazovic, 2015-09-15:
wow. thanks a lot.

bit confused. This code is in ofxOMXPlayer. Will this cause whole display
to be rotated or just a one ofxOMXPlayer instance ?

Nenad

PTS93, 2015-09-15:
Oh thats great to know. Maybe I find the time to break that out as an option.

@nenadlazovic 
This only effects video played back with ofxOMXPlayer.

nenadlazovic, 2015-09-17:
Did as proposed. Thanks.
I also looked at https://github.com/popcornmix/omxplayer/blob/master/OMXVideo.cpp.

I was not quite sure about meaning of your comment 
> Make sure configureDisplay(); by editing the code here
> https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/OMXVideo.cpp#L349
Please explain.

Below is modified configureDisplay() method. I am planning to add orientation to ofxOMXPlayerSettings in order to pass to to COMXVideo. 
Video performance portrait mode is quite decent. I tried 3 videos and it was playing quite fine (except that i might not have set x,y,w,h correctly).

As I am not familiar with ofxOMXPlayer code, would it be possible for you to add ofxOMXPlayerSettings.orientation attribute support ?

```
    ofxOMXPlayerSettings settings;
    ...
    settings.orientation = true;    //0,90,180,270  <--- new attribute
```
Modified **COMXVideo::configureDisplay()** method
```
void COMXVideo::configureDisplay()
{
        OMX_CONFIG_DISPLAYREGIONTYPE configDisplay;
        OMX_INIT_STRUCTURE(configDisplay);
        configDisplay.nPortIndex = m_omx_render.GetInputPort();

        configDisplay.set = (OMX_DISPLAYSETTYPE)(OMX_DISPLAY_SET_DEST_RECT|OMX_DISPLAY_SET_SRC_RECT|OMX_DISPLAY_SET_FULLSCREEN|OMX_DISPLAY_SET_NOASPECT | OMX_DISPLAY_SET_TRANSFORM);

        configDisplay.dest_rect.x_offset  = displayRect.x;
        configDisplay.dest_rect.y_offset  = displayRect.y;
        configDisplay.dest_rect.width     = displayRect.getWidth();
        configDisplay.dest_rect.height    = displayRect.getHeight();

        configDisplay.fullscreen = OMX_FALSE;
        configDisplay.noaspect   = OMX_TRUE;
        configDisplay.transform  = OMX_DISPLAY_ROT270;

        m_omx_render.SetConfig(OMX_IndexConfigDisplayRegion, &configDisplay);
}		
```

nenadlazovic, 2015-09-20:
managed to complete. works ok, using code above. thanks a lot.

closed 2015-09-20 by nenadlazovic