jvcleave/ofxOMXPlayer #60: some videos jitters first few seconds ...

nenadlazovic, 2015-09-20:
Hi jvcleave,

I have issue with some videos I am trying to play. 
Some videos plays ok all the way through. Some have jitter first few . 
I did few experiments. To avoid potential problems introduced by my app, I am using ofxOMXPlayer examples (example-basic, example-restartMovie, example-playlist, …)
I am using micro SD (class 10 UHS-I, per mfg “Up to 50MB/s”).
Video info for videos I am having issue with is below questions.

**Notes (when playing vide that experience jitter)**
* playing video have jitter first 10 sec
* after initial jitter period, video plays smoothly
* if settings.enableLooping = true, jitter’s only on first loop, subsequent loops have no jitter
* restarting video (via seekToTimeInSeconds(0) or restartMovie()) cause initial loop jitter
* similar problems exist in textured and non-textured mode
* while jitter’s CPU jumps to 30-40%. After jitter period, goes down to 5-10%
* same video doesnt have jitter when played in https://github.com/popcornmix/omxplayer

**Questions:**
1. is there way to preload video, and start play on demand ?
2. is there a way to restart video without causing re-loading (and consequently avoiding jitter) ?

**Video info** for video 1 that experience jitter
General
Complete name                  : C:\Users\nlazovic\Downloads\file-samples\GSKPanadolExtra.mp4
Format                         : MPEG-4
Format profile                 : Base Media / Version 2
Codec ID                       : mp42
File size                      : 20.8 MiB
Duration                       : 16s 600ms
Overall bit rate               : 10.5 Mbps
Encoded date                   : UTC 2015-09-13 22:52:19
Tagged date                    : UTC 2015-09-13 22:52:20
©TIM                           : 00:00:00:00
©TSC                           : 25
©TSZ                           : 1

Video
ID                             : 1
Format                         : AVC
Format/Info                    : Advanced Video Codec
Format profile                 : High@L5.1
Format settings, CABAC         : Yes
Format settings, ReFrames      : 3 frames
Codec ID                       : avc1
Codec ID/Info                  : Advanced Video Coding
Duration                       : 16s 600ms
Bit rate                       : 10.5 Mbps
Width                          : 1 080 pixels
Height                         : 1 920 pixels
Display aspect ratio           : 0.563
Frame rate mode                : Constant
Frame rate                     : 25.000 fps
Standard                       : PAL
Color space                    : YUV
Chroma subsampling             : 4:2:0
Bit depth                      : 8 bits
Scan type                      : Progressive
Bits/(Pixel*Frame)             : 0.203
Stream size                    : 20.8 MiB (100%)
Language                       : English
Encoded date                   : UTC 2015-09-13 22:52:19
Tagged date                    : UTC 2015-09-13 22:52:19
Color primaries                : BT.709
Transfer characteristics       : BT.709
Matrix coefficients            : BT.709

**Video info** for video 2 that experience jitter
General
Complete name                  : C:\Users\nlazovic\Downloads\file-samples\HF_FEBRICET.mp4
Format                         : MPEG-4
Format profile                 : Base Media / Version 2
Codec ID                       : mp42
File size                      : 30.0 MiB
Duration                       : 23s 840ms
Overall bit rate               : 10.6 Mbps
Encoded date                   : UTC 2015-09-13 15:03:11
Tagged date                    : UTC 2015-09-13 15:03:11
©TIM                           : 00:00:00:00
©TSC                           : 25
©TSZ                           : 1

Video
ID                             : 1
Format                         : AVC
Format/Info                    : Advanced Video Codec
Format profile                 : High@L5.1
Format settings, CABAC         : Yes
Format settings, ReFrames      : 3 frames
Codec ID                       : avc1
Codec ID/Info                  : Advanced Video Coding
Duration                       : 23s 840ms
Bit rate                       : 10.6 Mbps
Width                          : 1 080 pixels
Height                         : 1 920 pixels
Display aspect ratio           : 0.563
Frame rate mode                : Constant
Frame rate                     : 25.000 fps
Standard                       : PAL
Color space                    : YUV
Chroma subsampling             : 4:2:0
Bit depth                      : 8 bits
Scan type                      : Progressive
Bits/(Pixel*Frame)             : 0.204
Stream size                    : 30.0 MiB (100%)
Language                       : English
Encoded date                   : UTC 2015-09-13 15:03:11
Tagged date                    : UTC 2015-09-13 15:03:11
Color primaries                : BT.709
Transfer characteristics       : BT.709
Matrix coefficients            : BT.709

jvcleave, 2015-09-21:
Do the videos have audio?

I would try getting `Format settings, ReFrames : 3 frames` to be 1 frame

nenadlazovic, 2015-09-23:
we did reformat video and it is playing ok now ... will investigate more. anyhow thanks for suggestion.

closed 2015-09-23 by nenadlazovic

netdoggy, 2015-12-29:
Have the same problem.
Maybe there's another way to solve this problem without changing ReFrames video?
omxplayer plays correctly. The problem is similar as the author of this issue.

netdoggy, 2016-01-02:
@jvcleave, I was trying to convert video at 720p through ffmpeg also tried to change the reframes, but jitter still occur at the beginning. But it is not in omxplayer. You know why?

jvcleave, 2016-01-03:
@netdoggy is it just happening on loop?

netdoggy, 2016-01-03:
@jvcleave I tried various examples of "sample- *". I only change the video source to another video file. In all examples, there is a problem in the first few seconds, the first playback, if loop. I tested at Pi2.
I can send a small clip for your test, if it will help find a problem/solution.