jvcleave/ofxOMXPlayer #61: Unable to listen to onVideoEnd/onVideoLoop when not using ofApp class

PTS93, 2015-10-15:
I have written a custom class that wraps some of the elements of ofxOMXPlayer so I can abstract it and treat it like other types of media that I'm using in my application.

Now everything works fine so far I can play video just fine but even though I extended my class with ofxOMXPlayerListener my onVideoEnd, never triggers for unkown reasons.
I'm not exactly a pro when it comes to class inheritance and what not in C++. What am I doing wrong or is this a bug? When I had all my OMX related code in my ofApp class everything worked fine.

Thats how part of my class looks like (its fairly long), the Medium class is just an interface and doesn't do anything.
It also doesn't matter where I put my onVideo(..) functions, even if they are in ofApp and I extend ofApp with ofxOMXPlayerListener they don't get called.

``` C++
class VideoOMX: public Medium, public ofxOMXPlayerListener {
public:
	//lots of stuff
private:
	ofxOMXPlayerSettings settings;
	ofxOMXPlayer player;
	void onVideoEnd(ofxOMXPlayerListenerEventData& e){
		ofLogNotice("onVideoEnd");
	}
	void onVideoLoop(ofxOMXPlayerListenerEventData& e){
		ofLogNotice("onVideoLoop");
	}
}
```

jvcleave, 2015-10-15:
don't forget to set the listener 

https://github.com/jvcleave/ofxOMXPlayer/blob/master/example-shader/src/shaderApp.cpp#L30

PTS93, 2015-10-15:
Ugh yes, of course! Thanks so much!
And I thought I'm going slowly insane..

closed 2015-10-15 by PTS93