jvcleave/ofxOMXPlayer #63: no audio output - [error] COMXAudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring

lunchboxav, 2015-10-21:
Hi,

I encountered this error message when playing my video and I didn't hear any audio, though I've only compiled the example-basic app

[error] COMXAudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring

I tried the gist here https://gist.github.com/jvcleave/4972661 but that doesn't solve the problem either. 

Do you have any suggestion how to playback video with audio? I use my headphone jack btw.

lunchboxav, 2015-10-21:
apparently, I just need to disable audio from HDMI. sorry for this. consider this issue non exists.

closed 2015-10-21 by lunchboxav

vberthiaume, 2016-04-18:
I just ran into this issue, and I need the sound to be sent through HDMI... I can't seem to reopen this issue, should I create a new one?

jvcleave, 2016-04-18:
@vberthiaume did you try the gist above? Does omxplayer play through the HDMI?

vberthiaume, 2016-04-18:
Ah, sorry I hadn't seen the gist.  Will try and report.