jvcleave/ofxOMXPlayer #68: ofv0.9

mcanet, 2015-12-16:
If I create a project with project generator adding the addon get following error that happens in the linkage time:

/home/pi/openFrameworks/addons/obj/linuxarmv7l/Release/ofxOMXPlayer/src/OMXAudioDecoder.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status

The examples of the addon compile good.

jvcleave, 2015-12-16:
On what platform are you running the PG? Typically for the RPI you don't use it.

mcanet, 2015-12-16:
I am using raspbian Jessie and raspberry pi 2. And I call in the terminal project generator: ./projectGenerator -o”../../../..” -a”ofxOpenCv,ofxCv,ofxCvPiCam” -p”linux” ../../../../apps/test/player

jvcleave, 2015-12-16:
have you tried `make clean`?

jvcleave, 2015-12-16:
Your line didn't include this addon but I tried the below and it worked for me

`./projectGenerator -o"../../../.." -a"ofxOMXPlayer" -p"linux" ../../../../apps/test/player`

mcanet, 2015-12-16:
Sorry. this week I am not so many sleep. 

I saw some issue between ofxOMXPlayer and ofxRPiCameraVideoGrabber. Because they have a header duplicated OMX_Maps.h. But finally i end up going with ofxCvPiCam.

You did great work with the OF addons for Raspberry PI. ;-)

closed 2015-12-31 by jvcleave