jvcleave/ofxOMXPlayer #69: Opening http stream sometimes causes segmentation fault.

mojibaked, 2015-12-29:
So i'm trying to open a live video stream from a "master" Raspberry PI through LAN, like this:
```
ofxOMXPlayerSettings settings;
settings.videoPath = "http://192.168.1.20:8080";
```
7/10 times this works flawlessly, but for the rest i'm getting a segfault.
```
[notice ] bool OMXReader::open(std::string, bool): avformat_find_stream_info TOOK 5003 MS
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 5137 MS
[warning] bool ofxOMXPlayerEngine::setup(ofxOMXPlayerSettings&): FAST PATH MOVE OPEN FAILED - LIKELY A STREAM, TRYING SLOW PATH
Segmentation fault
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:168: recipe for target 'run' failed
make: *** [run] Error 139
```

I'm trying to edit ofxOMXPlayerEngine to reconnect after failing to open the stream the first time.
Not getting anywhere so far. I want the application to listen for the stream, and connect whenever it is ready. Even when the master sends out no video for a second. 
Whenever i try to call the `didOpenMovie = didReadFile(doSkipAvProbe);` or `didOpenMovie = omxReader.open(moviePath.c_str(), doSkipAvProbe);` again, i also get a segmentation fault.
Not sure if i need to clean up variables set inside omxReader.open() when it is executed the first time. Or if i'm staring at the wrong piece of code.

When the current video sent out by the master stops, and the next video in the playlist is up, the application crashes too. Preferably the application should not crash, and just play the next video..

Can someone please enlighten me?

jvcleave, 2015-12-29:
It sounds like libav can't detect the format of the stream

mojibaked, 2015-12-30:
Thanks for your reply!
I was sending out a live 720p stream from the master with vlc. Syncing multiple raspberry's using nginx/ffmpeg and rtp or rtsp caused more problems at that time (tearing, latency).
Changing mux=ts to mux=raw -seems- to have fixed the problem for now. In this line:
`cvlc -vvv video.mp4 -L -sout '#standard{access=http,mux=ts,dst=192.168.1.20:8080/stream}'`
Still have to figure out what to do when the stream loops.