jvcleave/ofxOMXPlayer #70: play only sound in texture mode

test23test, 2016-01-31:
Hi.
I’m trying to example-basic compile but play only sound .
In non-texture mode, movie and sound can play.
How can I do play movie and sound in texture mode?
What's the cause?
I tested some other example, but all the same.

I'm on Raspberry Pi2 using the latest Jessie.
oF0.9.0
[https://github.com/jvcleave/ofxOMXPlayer/releases/tag/0.9.0-compatible](url)

```
pi@raspberrypi:~/openFrameworks/apps/myApps/example-basic $ make run
HOST_OS=Linux
HOST_ARCH=armv7l
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl libpulse-simple alsa gtk+-3.0 
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1080
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): mouse_fd= 5 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.3:1.0-event-mouse
[notice ] ofAppEGLWindow: setupMouse(): mouse device name = PixArt Microsoft USB Optical Mouse
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 6 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.5:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name = HID 0566:3029
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 8 MS
[ error ] COMXAudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 30.0417
[notice ] bool OMXClock::OMXStart(double, bool): at pts: 0
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[ error ] OMX_ERRORTYPE COMXCoreComponent::WaitForCommand(OMX_COMMANDTYPE, OMX_U32, long int): OMX.broadcom.video_decode OMX_CommandStateSet
[notice ] ofAppEGLWindow: destroyNativeKeyboard()
[notice ] ofAppEGLWindow: destroySurface(): destroying EGL surface
[notice ] ofAppEGLWindow: display(): eglSwapBuffers failed: invalid display specified
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:168: recipe for target 'run' failed
make: *** [run]
```

PTS93, 2016-01-31:
Make sure you allocated atleast 256MB RAM for the GPU. Use `sudo raspi-config`.
https://www.raspberrypi.org/documentation/configuration/raspi-config.md

test23test, 2016-01-31:
Thanks.
I rewrote RAM to 256MB but nothing changes.

test23test, 2016-01-31:
I rewrote follows, finally I made it!
Thanks.

```
//so either pass in the settings
//omxPlayer.setup(settings);
//or live with the defaults
omxPlayer.loadMovie(videoPath);
```
to

```
//so either pass in the settings
omxPlayer.setup(settings);
//or live with the defaults
//omxPlayer.loadMovie(videoPath);
```

closed 2016-01-31 by test23test