jvcleave/ofxOMXPlayer #72: CRASHING HERE219

bigfatbrowncat, 2016-02-28:
I have a problem with this sample video: http://www.sample-videos.com/video/mp4/720/big_buck_bunny_720p_50mb.mp4

    [ error ] bool OMXAudioPlayer::decode(OMXPacket*): CRASHING HERE219

Your included version ("Timecoded_Big_bunny_1.mov") works perfect.

Is it problem with decoder? With memory? I have no idea, could you give a tip?

jvcleave, 2016-02-28:
Try encoding the audio as 48k. PCM also usually works pretty well

bigfatbrowncat, 2016-02-28:
Is there any way to check if the audio stream is supported or not before the app crashes? :)

jvcleave, 2016-02-28:
you can remove _Exit(0) line but I probably did that to stop the pi from
locking up

bigfatbrowncat, 2016-02-28:
I just want it to report "I can't play video" as a callback or something instead of locking, hanging or halt. Is it possible?
I am trying to create a remotely-controlled video player and it's not good for it to crash on some unsupported video/audio formats.

jvcleave, 2016-02-28:
Not sure. It's a condition that I don't understand (the audio codec gets changed mid-play?). I'll take a look at your video and see if I can replicate.

jvcleave, 2016-02-29:
Not sure what is causing it yet but just updated master with bb2b96642eed0cd710ff67d72d175da839888fce so it doesn't exit. Be aware this may cause leaks further down the road. 

It also doesn't have a way of calling you back yet