jvcleave/ofxOMXPlayer #77: ofxOMXPlayer Freezes

DHaylock, 2016-03-31:
Hi @jvcleave,

I'm having two really unusual issues.
One is where the video freezes, but the current frame count keeps progressing. 
Second is similar but the current frame freezes. 
I've tried all the examples and they all seem to fall over with either issue.
Tried updating the Pi. Nothing
Cleaned the applications. Nothing

I'm running Jessie 4.1.20-v7*
Running of_0.9.0_linuxarmv7l
Memory is Split: 256 GPU
Overclocked: High

Only error log i'm getting is the following.

Any ideas?

<pre>[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1080
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): unabled to find mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse, mouse_fd < 0
[warning] ofAppEGLWindow: setupKeyboard(): unabled to find keyboard
[ error ] ofAppEGLWindow: setupKeyboard(): did not open keyboard
[ error ] ofAppEGLWindow: setupKeyboard(): did not open keyboard, keyboard_fd < 0
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 43 MS
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] AudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 33.3333
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[ error ] OMX_ERRORTYPE Component::waitForEvent(OMX_EVENTTYPE, long int): OMX.broadcom.audio_render waitForEvent Event: OMX_EventPortSettingsChanged TIMED OUT at: 20</pre>

jvcleave, 2016-03-31:
hard to say without seeing the video

DHaylock, 2016-04-04:
Hi @jvcleave, 

https://vimeo.com/161448026

Best

jvcleave, 2016-04-04:
I mis-read it as you were having issues with your own video. Anyway the video helped :)

Can you try seeing if you have the same issue with your monitor set to 720p?

I set it like this in /boot/config.txt

```
# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
#720p
hdmi_mode=4
#1080p
#hdmi_mode=16
```

DHaylock, 2016-04-04:
Hi @jvcleave 

No worries.
Yep same error.
Tried running cmdline omxplayer and was pretty much fine, think it dropped a frame.
Converted the Video to be exactly the same as suggested in the ReadMe.

Its this error thats weirding me out.

<pre> [ error ] static OMX_ERRORTYPE Component::EventHandlerCallback(OMX_HANDLETYPE, OMX_PTR, OMX_EVENTTYPE, OMX_U32, OMX_U32, OMX_PTR): 1072 OMX_ErrorSameState
</pre>

Best

jvcleave, 2016-04-04:
that error is fine - it just means something tried to send an component to another state and it was already there

I just compiled master and ran example-basic on an RPI2 and it was ok

here is my entire config.txt 

```
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
#720p
hdmi_mode=4
#1080p
#hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=950

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

start_x=1
gpu_mem=256
core_freq=250
sdram_freq=450
over_voltage=6
dtparam=spi=on
dtparam=i2c_arm=on

```

DHaylock, 2016-04-04:
Cheers @jvcleave, 

I'll try them out

Best

zetteeh, 2018-07-06:
hey, 
any progress on the issue? 
got the same problems on a RP 3 b+ model. 

the video freezes. then i can only exit via `kill`. 

thanks!

```
[notice ] MediaServer: Removing video /home/pi/media/00-FIRST.mp4
[notice ] MediaServer: Source count before video removal: 1
[notice ] MediaServer: Source count after video removal: 0
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 139 MS
[ error ] AudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 907.16
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[notice ] MediaServer: Initialized reference count of /home/pi/media/01-SECOND.mp4 to 1
[ error ] OMX_ERRORTYPE Component::waitForEvent(OMX_EVENTTYPE, long int): OMX.broadcom.audio_render waitForEvent Event: OMX_EventPortSettingsChanged TIMED OUT at: 20
[ error ] OMX_ERRORTYPE Component::setParameter(OMX_INDEXTYPE, OMX_PTR): 914 OMX_ErrorBadParameter
[ error ] unsigned int OMXAudioDecoder::addPackets(void*, unsigned int, double, double): 828 OMX_ErrorBadParameter
[ error ] OMX_BUFFERHEADERTYPE* Component::getInputBuffer(long int): OMX.broadcom.video_decode TIMEOUT
[ error ] bool BaseVideoDecoder::decode(uint8_t*, int, double): Decode timeout
```