jvcleave/ofxOMXPlayer #79: -lsmbclient and -lssh

pikilipita, 2016-04-07:
I just updated oofxomxplayer ot the latest sources.
When compiling, I'm hitting this error:
cannot find -lsmbclient 
cannot -lssh

Any suggestions?

jvcleave, 2016-04-07:
yeah - I recompiled ffmpeg in trying to see if I could get optimizations. I will probably recompile without the smb/ssh requirements but until then this should work

`sudo apt-get install libsmbclient-dev libssh-dev`

pikilipita, 2016-04-07:
Thanks, will try that

jvcleave, 2016-04-08:
btw - I've started including the Makefiles for ffmpeg if you want to do it yourself
e.g
https://github.com/jvcleave/ofxOMXPlayer/blob/master/libs/linuxarmv6l/Makefile

jvcleave, 2016-05-25:
fixed for arm6 here:
a50b7fc5f4dc987f9ee652dc90d535207d4a66cd

closed 2016-09-14 by jvcleave