jvcleave/ofxOMXPlayer #80: Problems playing Full HD video

antasa, 2016-04-09:
I have an issue with playing full HD videos the ofxOMXplayer. We are using openframeworks 0.8.4. The video we are playing is not smooth, the playback is jumpy and jittery. It seem to drop frames. The video is 1920x1080 @30fps.The frame rate is constant and OK, over 50fps so that is not the issue. We can also successfully run same video in omxplayer directly, the issue only occurs when we run it with the ofxOMXplayer. The video is the format suggested. mpegstreamclip at 50% quality and with a soundtrack.
We can run a scaled down version 720p successfully on ofxomxplayer. The issue is only when we use the Full HD version. Do you have any suggestions we could try to get it play full HD smoothly.

antasa, 2016-04-09:
Sorry forgot to say. It's not in textured mode. The issue is also there in Normal mode

jvcleave, 2016-04-09:
Can you provide more detail?

1. RPi version (e.g RPi1, RPi2 or RPi3)
2. Memory split? (e.g. 384 to GPU)
3. Screen resolution/App resolution
4. Link to video? (can email to jvcleave@gmail.com)

thomashollier, 2016-04-09:
Antasa and I are working on this together and I can add the following detail. 

The app resolution is 1280x720 but we are doing full screen playback through the ofSetFullscreen(...) function, which is scaling it to the full screen. Our HDMI screen is 1680x1050 and the movie is 1920x1080.

I guess as I write this, I'm wondering if all these various resolutions may not be causing our problem. Could it be the ofSetFullScreen() causing the problem?

jvcleave, 2016-04-09:
not sure - per the email I tried it with the latest master and it was fine with the exception of some glitching that I was able to resolve by recompressing it with more keyframes

antasa, 2016-04-11:
Hi, thanks for the support and suggestions. We figured out that we had textured mode set to enabled in our App, so this is why the videos where not playing back smoothly. I can confirm the video runs smoothly in full HD.

closed 2016-04-11 by antasa