jvcleave/ofxOMXPlayer #81: Do clean exit when sending Ctrl+C (SIGINT)

0xC0DEBA5E, 2016-05-25:
When I exit the app via "ctrl+c" over a SSH-Connection, a SIGINT gets sent to the process. Sometimes the App crashes with a segfault and the process is still stuck in the process-list (ps aux | grep <program name>). All further attempts to kill the hanging process were unsuccessful and i need to power-cycle the RPi, otherwise sometimes the GPU is still doing some weird stuff from the previously crashed process.

The hanging process doesnt react to any signals anymore. Maybe the `void ofxOMXPlayer::addExitHandler()`-method doesn't work as expected?

Does `ofLogVerbose(__func__)` << " EXITING VIA SIGNAL";` get print to stdout or stderr when I run with `make RunRelease`? Because I think I should see this, if the `signal_handler'-method is beeing called.

jvcleave, 2016-05-25:
this happening with the examples or just your app?

0xC0DEBA5E, 2016-05-25:
I will try it with example-basic right now.

PTS93, 2016-05-25:
I can verify that. Happens to me a lot, not sure if its related to ofxOMXPlayer
though.

jvcleave, 2016-05-25:
The thing to look I use is `ps axv` and look the `STAT` column - if this has a Z by your app than it is usually a hardware(GPU) problem often related to GPU memory split

A couple of other things

```
sudo vcdbg reloc | grep "free memory"
sudo vcdbg reloc | grep Orphaned
```

0xC0DEBA5E, 2016-05-25:
It doesn't happen with the video which is included in the example. 
It crashed with the following config:

- Memory-Split: 256MB
- Video
    - Resolution: 1280*800
    - Encoding: PhotoJPEG 25FPS

Sometimes the process crashes with a segmentation fault and sometimes it just freezes. When a segfault occurs, not all (maybe none) GPU-Memory is freed.

0xC0DEBA5E, 2016-05-26:
I think the destructor of the videoPlayer is not beeing called in the destructor in ofxOMXPlayerEngine.cpp on line 738? videoPlayer is only set to NULL instead of deleting it. Therefore `void VideoPlayerTextured::close()` doesnt get executed. I dont know how important that is, but it should maybe get fixed.

Have you considered using unique_ptr instead of raw pointers? With them you can get rid of errors like these with zero overhead (you dont have to call delete manually).

jvcleave, 2016-05-26:
I was able to run the video and close without any segfaults/closing issues 

RPi2 with 256 GPU memory

Here is my config.txt in case it helps:

```
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
#720p
hdmi_mode=4
#1080p
#hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=950

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

start_x=1
gpu_mem=256
core_freq=250
sdram_freq=450
over_voltage=6
dtparam=spi=on
dtparam=i2c_arm=on
```