jvcleave/ofxOMXPlayer #82: Decode timeout on Raspberry pi 2

kurmanovd, 2016-06-02:
Excuse my English.

raspberry pi 2
Linux 3.18.13-v7+ #784 SMP PREEMPT Sat May 9 15:57:36 BST 2015 armv7l GNU/Linux
Running of_0.9.3_linuxarmv6l
Memory is Split: 384 GPU

When I start "example-texture-mode" project with the file [.mp4](https://www.wetransfer.com/downloads/15e79a543cbf2fbec5011b5fc5efea8b20160602083242/449a2fc8ec805dbd793a0c4e08f7635e20160602083242/50cd77) constantly I receive an error of "OMX.broadcom.video_decode TIMEOUT".

> [notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1280x720
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): unabled to find mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse
[ error ] ofAppEGLWindow: setupMouse(): did not open mouse, mouse_fd < 0
[warning] ofAppEGLWindow: setupKeyboard(): unabled to find keyboard
[ error ] ofAppEGLWindow: setupKeyboard(): did not open keyboard
[ error ] ofAppEGLWindow: setupKeyboard(): did not open keyboard, keyboard_fd < 0
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 52 MS
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 12.16
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[ error ] Component::~Component(): 80 OMX_ErrorBadParameter
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 12 MS
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 71.56
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[ error ] OMX_BUFFERHEADERTYPE* Component::getInputBuffer(long int): OMX.broadcom.video_decode TIMEOUT
[ error ] bool BaseVideoDecoder::decode(uint8_t*, int, double): Decode timeout
[ error ] OMX_BUFFERHEADERTYPE* Component::getInputBuffer(long int): OMX.broadcom.video_decode TIMEOUT
[ error ] bool BaseVideoDecoder::decode(uint8_t*, int, double): Decode timeout
[ error ] OMX_BUFFERHEADERTYPE* Component::getInputBuffer(long int): OMX.broadcom.video_decode TIMEOUT
[ error ] bool BaseVideoDecoder::decode(uint8_t*, int, double): Decode timeout 

I tried to change `OMX_BUFFERHEADERTYPE value * to omxBuffer = decoderComponent.getInputBuffer(500);` on 1000 and 2000ms, but then after a while reproduction absolutely stops and I continuously receive messages of "OMX.broadcom.video_decode TIMEOUT"

jvcleave, 2016-06-02:
it's typical - nothing to worry about unless you have other issues

kurmanovd, 2016-06-02:
Thanks for the answer!
The problem is that during the work 24х7, several hours later, the error of "Decode timeout" begins to appear continuously and reproduction stops...

jvcleave, 2016-06-02:
This happens with example-texture-mode?

kurmanovd, 2016-06-29:
apt-get update + apt-get upgrade - it seems have solved my problem

closed 2016-06-29 by kurmanovd

yderidde, 2016-08-28:
I run into this exact problem with a Raspberry Pi 2 Model B  .. I did apt-get update + apt-get upgrade and it didn't solve the problem.
What else can it be ? please help :)