jvcleave/ofxOMXPlayer #85: Implement "Stop" action.

MindContact, 2016-08-12:
Hi jvcleave, I'm trying your addon (release for oF090) with a rPi2/jessie/oF093 and A/V playback it's ok.
I need a stop/play methods, how I can implement these?

Thanks in advance!

jvcleave, 2016-08-12:
it depends on what you expect them to do really. `setPaused(true)` is essentially stop and you can use `loadMovie` to replay. There is also a `restartMovie`

take a look at example-playlist and example-playback-controls

MindContact, 2016-08-16:
Hello jvcleave,
I followed your advice, in the version 090 was not present example-playback-controls, present instead in the master branch.

Use a bool variable to know if omxPlayer stopped, as if you set setPaused (true) the isPlaying property remains true.

Finally, this variable is used in the draw () method for presenting video content only when it is in play.

Thanks again, see you soon!

---

void ofApp::video_Play(){
    if (video_playRequested && (!omxPlayer.isPlaying() || omxPlayer.isPaused())) {
        omxPlayer.restartMovie();
        video_playRequested = false;
        video_isPlaying = true;
    }
}

void ofApp::video_Stop(){
    if (video_stopRequested && (omxPlayer.isPlaying() || !omxPlayer.isPaused())) {
        omxPlayer.setPaused(true);
        video_stopRequested = false;
        video_isPlaying = false;
    }
}

void ofApp::setupMedia()
{
    // MEDIA ITEMS SETUP
    string videoPath = ofToDataPath(video_fileName, true);
    
    //Somewhat like ofFboSettings we may have a lot of options so this is the current model
    ofxOMXPlayerSettings settings;
    settings.videoPath = videoPath;
    settings.useHDMIForAudio = false;	//default true
    settings.enableTexture = true;		//default true
    settings.enableLooping = false;		//default true
    settings.enableAudio = true;		//default true, save resources by disabling
    //settings.doFlipTexture = true;		//default false

    //so either pass in the settings
    omxPlayer.setup(settings);
    video_Stop();
}

void ofApp::update(){
    if (video_playRequested) {
        video_Play();
    }
    
    if (video_stopRequested) {
        video_Stop();
    }
}

void ofApp::draw(){
    if (video_isPlaying)
        omxPlayer.draw(playerVideoLeft, 0, videoW, videoH);
}

closed 2016-08-16 by MindContact