jvcleave/ofxOMXPlayer #87: onVideoEnd sometimes does not fire

adamkalicinscy, 2016-08-25:
I have playlsit with 2 videos (30fps h264) with endless loop loading in non-texture mode.  Sometimes one video ends and onVideoEnd does not file and next video can not load. 

There is no error and it happens randomly. Sometimes after 3 loops sometimes after 100.

I have Raspberry Pi 3 with latest wheezy

adamkalicinscy, 2016-08-25:
I tested it on files from the phone, which had a variable frame rate (18.676 fps - 30.477 fps). It could be a problem.
Now I am testing files with a fixed number of frames, and it seems to be ok.

adamkalicinscy, 2016-08-27:
The problem still appears now after 7 hours and 280 video playbacks

jvcleave, 2016-08-27:
is it a particular movie it is failing on or different ones? 

You could try moving onVideoEnd() up to here
https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayerEngine.cpp#L307

adamkalicinscy, 2016-08-27:
Videos are `Timecoded_Big_bunny_1.mov` and http://pdl.warnerbros.com/wbmovies/gravity/trailer2/GRAVITY_TRAILER_5-1080.mov

I need a solution that is reliable and works non-stop 24 hours a day. For now I've done the dirty solution but reliable. I rely on the system time and length of the movie `getDurationInSeconds`. I will check your hint but unfortunately probably again I have to wait at least seven hours on the result :/

By the way, I discovered one more thing. For the film `Timecoded_Big_bunny_1.mov` `getCurrentFrame `at some point it is much higher than `getTotalNumFrames`. Someone has already had a problem with it https://github.com/Hemisphere-Project/HPlayer/issues/34

adamkalicinscy, 2016-08-29:
Now after 30h loop I have "Segmentation Fault" and app crashes :/

jvcleave, 2016-08-30:
Another thing to try (without the fix above)

I think the audio cache is getting an incorrect value/not decrementing to 0 at times. 

If you are not using audio, disabling it in ofxOMXPlayerSettings may work as well
```
            isCacheEmpty = !videoPlayer->getCached();
#if 0
            if (hasAudio)
            {
                if (!audioPlayer->getCached() && !videoPlayer->getCached())
                {
                    isCacheEmpty = true;
                }
            }
            else
            {
                if (!videoPlayer->getCached())
                {
                    isCacheEmpty = true;
                }
            }
#endif       
```

adamkalicinscy, 2016-08-31:
Ok I will try `enableAudio = false`. I had only `useHDMIForAudio = false `