jvcleave/ofxOMXPlayer #88: Decode Timeout on Raspberry Pi2 Model B

yderidde, 2016-08-28:
When i run `example-texture-mode` i get 

```
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): screenRect: 1920x1200
[notice ] ofAppEGLWindow: setupRPiNativeWindow(): windowRect: 1280x720
[notice ] ofAppEGLWindow: createSurface(): setting up EGL Display
[notice ] ofAppEGLWindow: createSurface(): EGL Display correctly set 0x1
[notice ] ofAppEGLWindow: createSurface(): no current renderer selected
[notice ] ofAppEGLWindow: createSurface(): default renderer detected
[notice ] ofAppEGLWindow: createSurface(): surface created correctly
[notice ] ofAppEGLWindow: createSurface(): API bound correctly
[notice ] ofAppEGLWindow: createSurface(): -----EGL-----
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MAJOR = 1
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION_MINOR = 4
[notice ] ofAppEGLWindow: createSurface(): EGL_CLIENT_APIS = OpenGL_ES OpenVG
[notice ] ofAppEGLWindow: createSurface(): EGL_VENDOR = Broadcom
[notice ] ofAppEGLWindow: createSurface(): EGL_VERSION = 1.4
[notice ] ofAppEGLWindow: createSurface(): EGL_EXTENSIONS = EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface 
[notice ] ofAppEGLWindow: createSurface(): GL_RENDERER = VideoCore IV HW
[notice ] ofAppEGLWindow: createSurface(): GL_VERSION  = OpenGL ES-CM 1.1
[notice ] ofAppEGLWindow: createSurface(): GL_VENDOR   = Broadcom
[notice ] ofAppEGLWindow: createSurface(): -------------
[notice ] ofAppEGLWindow: setupPeripherals(): peripheral setup complete
[notice ] ofAppEGLWindow: setupNativeUDev(): created udev object
[notice ] ofAppEGLWindow: setupMouse(): mouse_fd= 5 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.4:1.0-event-mouse
[notice ] ofAppEGLWindow: setupMouse(): mouse device name = PIXART USB OPTICAL MOUSE
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard_fd= 6 devicePath=/dev/input/by-path/platform-3f980000.usb-usb-0:1.5:1.0-event-kbd
[notice ] ofAppEGLWindow: setupKeyboard(): keyboard device name = Logitech USB Keyboard
[notice ] ofAppEGLWindow: setupPeripherals(): native event setup complete
[notice ] bool ofxOMXPlayerEngine::didReadFile(bool): didOpenMovie TOOK 18 MS
[ error ] AudioCodecOMX::GetChannelMap - FFmpeg reported 2 channels, but the layout contains 0 ignoring
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): duration SET: 33.3333
[notice ] bool ofxOMXPlayerEngine::openPlayer(int): Opened video PASS
[ error ] OMX_ERRORTYPE Component::waitForEvent(OMX_EVENTTYPE, long int): OMX.broadcom.audio_render waitForEvent Event: OMX_EventPortSettingsChanged TIMED OUT at: 20
[ error ] OMX_BUFFERHEADERTYPE* Component::getInputBuffer(long int): OMX.broadcom.video_decode TIMEOUT
[ error ] bool BaseVideoDecoder::decode(uint8_t*, int, double): Decode timeout
```

I know there was a previous issue that was closed but I can't seem to fix it with the proposed workaround/fix which is to run apt-get update / upgrade ...

jvcleave, 2016-08-28:
the message is just a warning

yderidde, 2016-08-28:
but the screen stay gray :/

jvcleave, 2016-08-28:
that usually means that you are out of GPU memory. you should have at least 128 allocated to the GPU in your memory split. 256 is better

yderidde, 2016-08-29:
Thanks it works with 256 :)

closed 2016-08-29 by yderidde