jvcleave/ofxOMXPlayer #89: Calling loadMovie() or setup() seems to freeze the system for a couple seconds on a raspberry pi2

yderidde, 2016-08-31:
I basically need to be able to load on the fly video files from a playlist on a user action (turn a physical dial where each step is another video) and would like to reduce the lag/unresponsiveness.

Right now when I call loadMovie() or setup() it seems to recreate a new ofxOMXPlayerEngine class and re-initialize everything and this apparently takes all resources and the app becomes unresponsive for 2-3 seconds on a Raspberry Pi 2 Model B

I was wondering why loadMovie() is recreating / re-initializing everything every time instead of just keeping the same context and settings and just switch the input content (video file) ?

- Is there any good reasons for this ?
- Is there a way to optimize the current implementation so the app stays responsive when switching video ?  If yes where you would recommend me to look at ? (I'm willing to try to optimize this ..)

jvcleave, 2016-08-31:
The way is is now (destroying the internal engine and reinitializing it) has been the easiest/fastest way I have found working with the OMX components. omxplayer does something similar

The biggest bottleneck I have found is when ffmpeg/libav probes the file to figure out what format it is.

I try a quick check and if that fails to get the required info it uses a slower probe (typically needed for streams)

https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayerEngine.cpp#L62

yderidde, 2016-08-31:
I just looked at the log and here it is definitely using the "fast path". It is in fact not a stream, it is local h.264 encoded movie at 720x480 size (using composite Video output).

The bottleneck seems to be elsewhere. After measuring it takes more like 3-5 seconds to load a movie and during that time the system is unresponsive which is not ideal when there are user interaction that are supposed to happen. 

Any other recommendation of where to look ?
Is there something that other people have experienced ?

jvcleave, 2016-08-31:
2-3 seconds is definitely slower than usual - the examples should demonstrate that.

i don't think composite out uses GPU/OMX acceleration - I'm actually surprised it works at all.

yderidde, 2016-08-31:
Do you have any documentation that prove that GPU/OMX acceleration  is not working with composite out ? that seems quite strange if it was the case. Why composite would not allow acceleration ?

yderidde, 2016-08-31:
I did setup 256MB for GPU for memory split.
I have a pretty basic Raspbian setup with a modified example-basic where you can switch video when pressing a keyboard key ...  If 3-5 seconds to load a movie using the "fast path" is not normal where do you think I should look ? Also I will right away run the same thru HDMI and see if it gives better result and let you know in few mins

jvcleave, 2016-08-31:
I don't have the reference - I mostly remember being surprised when I read it. 

example-playlist does the switching with a key `n`

yderidde, 2016-08-31:
Ok I found the bottleneck .. and it is actually not ofxOMXPlayer :D ...

What is really causing the problem is the ofSoundPlayer .. loadSound +play takes approx 3-4 seconds ! 
'''
audioPlayer.loadSound(path);
audioPlayer.play();
'''

Have you experienced that before ..  Now I wonder what I should use to load an audio file separately from the video.. Maybe you have some suggestion ?

Thanks a lot for your help really !

jvcleave, 2016-08-31:
Haven't messed with sound by itself that much actually. Good to know the player works with composite though!

Good luck!

closed 2016-08-31 by jvcleave