jvcleave/ofxOMXPlayer #91: Can't make any of the examples

samelie, 2016-09-09:
I successfully installed [omxplayer](https://github.com/popcornmix/omxplayer) and played a video using it.
Here is the end of the readout when I try and compile example-basic on my RP3
`g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:336: recipe for target '../../../addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/Tunnel.o' failed
make[1]: *** [../../../addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/Tunnel.o] Error 4
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxOMXPlayer-0.9.0-compatible/example-basic'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2`

jvcleave, 2016-09-09:
the compiler is running out of memory - if you are using any job options (e.g. `make -j4`) try it without it

samelie, 2016-12-07:
That was it! Thank you sir

closed 2016-12-07 by samelie