jvcleave/ofxOMXPlayer #93: Stutter when using ofSetFramerate to reduce the frequency

Gorgull, 2016-09-20:
Hi,

My current project displays a 720p25 video on a 720p50 display. 
I'm working in Texture mode to map the video on 2D geometry using a simple shader. 

I've just figured my video playback randomly stutters when I use ofSetFrameRate to match the video's FPS (ofSetFrameRate(25)), in order to minimise the number of rendered pixels / second to what's necessary.

Stutters don't always happen at startup, and it can stop as suddenly as it appeared but more frequently when the video loops. 

I've checked that each rendered frame displays the next frame of the video as expected by using
moviePlayer.getCurrentFrame(), and nothing wrong here. 

Default OF video player features an "update" method to process next frame, but ofxOMXPlayer doens't have such a mechanism and I don't know how it's handling it. 
Could someone clarify this for me?

More importantly, any idea why the stuttering, and how it could be fixed?

jvcleave, 2016-09-20:
the update is called automatically/internally. I didn't see a need to require the user to call (it's necessary internally and easily forgotten)

The easiest way to avoid stuttering is to add more keyframes in the video. 1:1 frame/keyframe ratio is best but will make your video file much larger.

Gorgull, 2016-09-21:
You mean a movie with 1 keyframe / frame isn't more VPU-intensive for the Raspberry Pi? 

Can I expect the same overall performance?

Gorgull, 2016-09-22:
I'd like to understand the way the update works in order to get rid of those weird stutters I get for no reason.
Do you call the update process synchronously? 
Is there any way I can control the update myself?

jvcleave, 2016-09-22:
https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayer.cpp#L39