jvcleave/ofxOMXPlayer #94: Volume changes back to initial value when seeking

0xC0DEBA5E, 2016-10-10:
When i change the volume during playback and do a seek afterwards, there is a sudden unwanted change in the volume.

I'm using 0.9.0 compatible and when I call `omxPlayer.seekToTimeInSec(x)`, the initial volume from the settings is loaded instead of the actual value. I think it is because `engine->openPlayer(startTimeInSeconds)` (ofxOMXPlayerEngine.cpp line 296) reinitializes the player with the ofxOMXPlayerSettings-object, in which the actual volume value isn't stored. Calls to `omxPlayer.setVolume()` don't change the volume in the settings-object.

jvcleave, 2016-10-10:
Thanks - I have a pretty big update coming up but if you are using pre-c++11 it may not include the fix. If you want to patch this for your instance I would just write the volume into settings before the engine restart

0xC0DEBA5E, 2016-10-11:
Here is what I did to fix this: I added the following at line 451  to `ofxOMXPlayer.cpp`: 
`settings.initialVolume = volume;`

So now the function `setVolume(float volume)` looks like this:

```C++
void ofxOMXPlayer::setVolume(float volume)
{
	if (engine)
	{
		engine->setVolume(volume);
		settings.initialVolume = volume;
	}
}`

jvcleave, 2016-10-11:
Looks good = you may also need to add for `ofxOMXPlayer::increaseVolume` and `ofxOMXPlayer::decreaseVolume`

`        settings.initialVolume = getVolume();`

0xC0DEBA5E, 2016-10-11:
Thanks, you were right. I added it to these lines too.

jvcleave, 2016-10-11:
actually I am surprised your above fix is working for you - I needed to do

```
void ofxOMXPlayer::setVolume(float volume)
{
    if (engine)
    {
        engine->setVolume(volume);
        engine->omxPlayerSettings.initialVolume = getVolume();

    }
}
```
and also `engine->omxPlayerSettings.initialVolume` for increase/decrease

0xC0DEBA5E, 2016-10-11:
I did not test the increase/decrease functionality, because I use a slider as a volume control. I am not on the latest development branch (I'm using 0.9.0-compatible). Maybe I can test it, if you want.

I have seen that there isn't a setting-object in ofxOMXPlayer.h anymore (changed in commit df51df0 "scrubbing").