jvcleave/ofxOMXPlayer #96: DBUS Support

magdesign, 2016-10-12:
Is there any chance to implement dbus support in ofxOmxplayer like they have in regular Omxplayer?

I need a "jump to position" and a "show current position" function for implementing omxplayer-sync (https://github.com/turingmachine/omxplayer-sync).

Would be pretty cool the be able to sync various players over network.

jvcleave, 2016-10-12:
As I understand it dbus is really just a way to communicate with the player via a keyboard/input device so I would leave that up to OF/the developer (I use/include the TerminalListener).

"jump to" would be nice but I have had mixed results with anything seeking. Better frame tracking (current position) for Direct Mode will be in an upcoming release. Texture mode current frame is pretty accurate.

magdesign, 2016-10-12:
Its a way to communicate with the player.
So if I can read out the current position of the "master player" and send it via TCP to a "slave player" then there would be the possibility to sync. If you can do it with frame tracking that would also be cool. 
BTW.: Popcornmix fixed a lot of the seek issues, it seems to work now.

So I wait for a the new release.... thanks.

magdesign, 2016-12-21:
Any news about a new release with bdus support?

jvcleave, 2016-12-21:
I still think it should sit above the player so I wouldn't add this functionality to this addon

magdesign, 2016-12-22:
Regular omxplayer can be controlled via dbus controls.
I'd like to control ofxOmxplayer also via dBus controls. This would open up a lot of possibilites, for example his problems https://github.com/jvcleave/ofxOMXPlayer/issues/99 could be solved since with dbus the GOP size of  keyframes does not matter anymore.

As you mentioned dbus should sit above the player, and you will not integrate into this addon the only possibility to control the video time would be "current frame Texture" mode?

PTS93, 2016-12-22:
How would a protocol like dbus help overcome limitations of the hardware decoder?
I have to agree with jvcleave, it doesn't make sense to add such a feature to the addon but rather develop something ontop of it.
Though having something like the seekToFrame function in the stable branch might help with such projects.