jvcleave/ofxOMXPlayer #97: <IL/OMX_Core.h> not found

nanu-c, 2016-11-19:
```
cat: obj/linux64/Release/.compiler_flags: Datei oder Verzeichnis nicht gefunden
Compiling /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp
g++ -c -O3 -march=native -mtune=native -DNDEBUG -Wall -std=c++14 -DGCC_HAS_REGEX -DOF_USING_GTK -DOF_USING_GTK -I../../../addons/ofxOMXPlayer/src -I../../../addons/ofxOMXPlayer/libs/ffmpeg/include -fPIC -U_FORTIFY_SOURCE -Wall -ftree-vectorize -Wno-deprecated-declarations -Wno-sign-compare -Wno-unknown-pragmas  -pthread -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/AL -I/usr/include/rtaudio -I/usr/include/alsa -I/usr/include/libdrm -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/mirclient -I/usr/include/mircommon -I/usr/include/mircookie -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I../../../libs/fmodex/include -I../../../libs/glm/include -I../../../libs/glm/include/glm -I../../../libs/glm/include/glm/detail -I../../../libs/glm/include/glm/gtc -I../../../libs/glm/include/glm/gtx -I../../../libs/json/include -I../../../libs/kiss/include -I../../../libs/tess2/include -I../../../libs/utf8/include -I../../../libs/utf8/include/utf8 -I../../../libs/openFrameworks -I../../../libs/openFrameworks/gl -I../../../libs/openFrameworks/3d -I../../../libs/openFrameworks/video -I../../../libs/openFrameworks/types -I../../../libs/openFrameworks/utils -I../../../libs/openFrameworks/math -I../../../libs/openFrameworks/events -I../../../libs/openFrameworks/sound -I../../../libs/openFrameworks/communication -I../../../libs/openFrameworks/app -I../../../libs/openFrameworks/graphics -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/example-basic.xcodeproj/xcshareddata -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/example-basic.xcodeproj/xcshareddata/xcschemes -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src -I../../../addons/ofxOMXPlayer/src -I../../../addons/ofxOMXPlayer/libs/ffmpeg/include -MMD -MP -MF obj/linux64/Release/src/ofApp.d -MT obj/linux64/Release/src/ofApp.o -o obj/linux64/Release/src/ofApp.o -c /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp
In file included from ../../../addons/ofxOMXPlayer/src/ofxOMXPlayer.h:4:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.h:4,
                 from /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp:1:
../../../addons/ofxOMXPlayer/src/OMXInitializer.h:3:25: fatal error: IL/OMX_Core.h: Datei oder Verzeichnis nicht gefunden
 #include <IL/OMX_Core.h>
cat: obj/linux64/Release/.compiler_flags: Datei oder Verzeichnis nicht gefunden
Compiling /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp
g++ -c -O3 -march=native -mtune=native -DNDEBUG -Wall -std=c++14 -DGCC_HAS_REGEX -DOF_USING_GTK -DOF_USING_GTK -I../../../addons/ofxOMXPlayer/src -I../../../addons/ofxOMXPlayer/libs/ffmpeg/include -fPIC -U_FORTIFY_SOURCE -Wall -ftree-vectorize -Wno-deprecated-declarations -Wno-sign-compare -Wno-unknown-pragmas  -pthread -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/AL -I/usr/include/rtaudio -I/usr/include/alsa -I/usr/include/libdrm -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/mirclient -I/usr/include/mircommon -I/usr/include/mircookie -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I../../../libs/fmodex/include -I../../../libs/glm/include -I../../../libs/glm/include/glm -I../../../libs/glm/include/glm/detail -I../../../libs/glm/include/glm/gtc -I../../../libs/glm/include/glm/gtx -I../../../libs/json/include -I../../../libs/kiss/include -I../../../libs/tess2/include -I../../../libs/utf8/include -I../../../libs/utf8/include/utf8 -I../../../libs/openFrameworks -I../../../libs/openFrameworks/gl -I../../../libs/openFrameworks/3d -I../../../libs/openFrameworks/video -I../../../libs/openFrameworks/types -I../../../libs/openFrameworks/utils -I../../../libs/openFrameworks/math -I../../../libs/openFrameworks/events -I../../../libs/openFrameworks/sound -I../../../libs/openFrameworks/communication -I../../../libs/openFrameworks/app -I../../../libs/openFrameworks/graphics -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/example-basic.xcodeproj/xcshareddata -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/example-basic.xcodeproj/xcshareddata/xcschemes -I/home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src -I../../../addons/ofxOMXPlayer/src -I../../../addons/ofxOMXPlayer/libs/ffmpeg/include -MMD -MP -MF obj/linux64/Release/src/ofApp.d -MT obj/linux64/Release/src/ofApp.o -o obj/linux64/Release/src/ofApp.o -c /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp
In file included from ../../../addons/ofxOMXPlayer/src/ofxOMXPlayer.h:4:0,
                 from /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.h:4,
                 from /home/nanu/visuals/openFrameworks/addons/ofxOMXPlayer/example-basic/src/ofApp.cpp:1:
../../../addons/ofxOMXPlayer/src/OMXInitializer.h:3:25: fatal error: IL/OMX_Core.h: Datei oder Verzeichnis nicht gefunden
 #include <IL/OMX_Core.h>

```

jvcleave, 2016-11-19:
it looks like you are using the linux64 version of openFrameworks. you need the arm6 version 

http://www.openframeworks.cc/setup/raspberrypi/

nanu-c, 2016-11-19:
You are right! gst-omx doesn't exist for linux64

closed 2016-11-19 by nanu-c