jvcleave/ofxOMXPlayer #98: Loop offset

smarques, 2016-12-11:
Hi and first of all thank you for ofxOMXPlayer.

I have a small clip that I am running in loop mode.
If I enable verbose logging I see at every loop the following:
[verbose] Loop offset :  2593.50
.....
[verbose] Loop offset :   3358.25

So noticing this number grows and grows I am wondering: will it cause an overflow if I leave the player going for a long time (say in an installation)?

Thank you in advance

jvcleave, 2016-12-11:
as I understand it max double is 1.79769e+308 so reboot it every few decades and you should be fine

smarques, 2016-12-11:
:) sorry if the question was dumb, I have another probably dumb question, but I will place it in a different issue.
Appreciate your patience :/

closed 2016-12-11 by smarques

jvcleave, 2016-12-11:
no problem - it's a valid question and didn't really know myself. I just looked up the max value of double to get an estimate