jvcleave/ofxOMXPlayer #99: seekToTimeInSeconds and looping

smarques, 2016-12-11:
I am a bit lost, I will try to explain what I am trying to do and maybe you can give me a hint.
What I need is to control video speed and direction using a sensor.
Just imagine one of those pre-cinema boxes where you would rotate a handle that would turn a sequence of images giving the illusion of animation.
I am trying to reproduce the effect with video.
I know I could use an ImageArrayLoader of some kind, but then I think on the raspberry I would have memory problems if the images are big, and it might not be very fluid.

My solution for now is to have two videos, one of which is the exact copy of the other, but reversed.
My idea was to have two players and switch from one or the other according to the direction.
Which works fine. I can set speed on each player and I am almost there.

My problem is, when the direction changes I need to move the reversed video to the corresponding frame where the forward video has stopped and start playing from there.
At this point perfect clock sync is not really important.
What would be nice would be to be able to do something like:

`backwardPlayer.setCurrentFrame(totalFrames - forwardPlayer.getCurrentFrame())`

Do you think that would be possibile? Or am I missing a more obvious solution?

Thank so much in advance

jvcleave, 2016-12-11:
Inherently the problem with seeking is the H264 container. H264 is highly compressed and typically only has keyframe markers every few frames or so. This makes frame accurate seeking either an approximation or CPU intensive (slow). This process is primarily performed by libav/ffmpeg.

Overall it seems the hardware acceleration (OpenMax) is really set up to take a stream and play it forward - this is why you will see the engine teardown and restart (seeking to a new/past time) as this is the fastest way to do it.

You can help the approximation process by adding keyframes to every frame of your video during compression.

jvcleave, 2016-12-11:
I was working on a seekToFrame method that is in the develop-filters branch

however that is the last time I have messed with it and was trying to solve some other issues and may have created new ones

https://github.com/jvcleave/ofxOMXPlayer/blob/develop-filters/src/ofxOMXPlayer.h#L57

smarques, 2016-12-11:
Thank you very much, I will look at the branch. For my purposes it doesn't have to be a perfect seek, approximate seek would be fine I believe. 
Thanks again.