jvcleave/ofxOMXPlayer #100: Disabling the player

itzaks, 2017-01-10:
Hello!

Wonderful work with the OMXPlayer. Really stoked to see my old PI1 B+ handling multiple videos plus shaders on a steady 60fps. 

I try to achieve switching visibility between multiple players in realtime. It seems the fastest way to do this is to keep them playing and control which one is being drawn (to my shader). But it seems as a waste of resources to have the ones not playing running in the background. Is it possible to disable / enable them frictionless? When I pause them I experience a bit of a delay upon playing them again.

jvcleave, 2017-01-10:
Not sure the best way to do it out-of-the-box. Pausing it should stop most of the processing but you may need to do some modifications - maybe write something that makes the threads sleep (if they are still running) on pause?

itzaks, 2017-01-11:
Okay, thanks for the quick response. I'll look into it :) On a little sidenote, is it possible to pass the stream from ofxOMXRecorder to ofxOMXPlayer without writing to disk, keeping it in memory?

jvcleave, 2017-01-13:
Not really as ofxOMXPlayer uses ffmpeg/libav to probe the file to get the width/height/codec. It may be possible but again you would have to modify it

itzaks, 2017-01-13:
I see, I'll hack around and see what I can do. Thanks for your help!

closed 2017-01-13 by itzaks