jvcleave/ofxOMXPlayer #101: Audio ALSA support

magdesign, 2017-01-12:
Since original omxplayer supports ALSA sound,
for example: omxplayer -o alsa:hw:1,0 /location/of/video.mp4

It would be amazing to get also ALSA support on ofxOMXPlayer.

PTS93, 2017-01-12:
Oh wow, I totally missed that they implemented that! Big +1 for adding alsa support for ofxOMXPlayer. By far the biggest issue I usually have.

magdesign, 2017-11-06:
I think this will never come :-(

jvcleave, 2017-11-06:
@magdesign yes - I've been on other non-RPi projects so it's unlikely to come from me.

jvcleave, 2018-09-29:
the way you can enable it is to use

`omxPlayer.engine.m_config_audio.device = "omx:alsa"; `

before calling omxPlayer.setup();

on my config is it currently causing errors and stopping video from playing but the component is passing. I don't know if I need other config tweaks in order to work as expected

PTS93, 2018-09-29:
Odd but would definitely be great to have!
Main use is having an i2s audio hat or a usb audio interface.

jvcleave, 2018-09-29:
ok - i do have a usb sound card so I can try that. It's probably just a matter of passing the correct hardware id

jvcleave, 2018-09-29:
it does work   - use the syntax

```
omxPlayer.engine.m_config_audio.device = "omx:alsa";
omxPlayer.engine.m_config_audio.subdevice = "hw:1,0";

//works but sounded weird with my config
//omxPlayer.engine.m_config_audio.subdevice = "default"; 
```

PTS93, 2018-09-29:
thats really great!

closed 2018-10-12 by jvcleave