jvcleave/ofxOMXPlayer #113: Reading from MJPEG and h264 streams

steeley, 2017-12-14:
Can this player read from a network stream of any sort? MJPEG from mjpeg-streamer, or h264 from FFMpeg would be great. I did try but it rejected the string I gave it.

thx.

PTS93, 2017-12-14:
See the stream example: https://github.com/jvcleave/ofxOMXPlayer/blob/master/example-http-stream/src/ofApp.cpp

RTSP is supported by ffmpeg: http://ffmpeg.org/ffmpeg-protocols.html#rtsp

steeley, 2017-12-14:
Thanks very much - I will give it a go.

steeley, 2017-12-14:
Unfortunately although it  basically works the performance is slow and jerky. 
The same videos play perfectly in VLC on the PI.  
ofxOMXplayer plays local video files in the examples ok.

hgshoggins, 2017-12-24:
Considering that "Majority of the code is based off of https://github.com/popcornmix/omxplayer" and that OMXPlayer can handle SDP files with streaming RTP H264, is it safe to assume that somehow one could feed an SDP file to ofxOMXPlayer and get it working ?

I'm asking this because I intend to use ofxPiMapper which depends on ofxOMXPlayer on Raspberry Pi, and I'm in need of such a feature : https://github.com/kr15h/ofxPiMapper/issues/69

PTS93, 2017-12-24:
Likely, try it out I guess. SDP is just a meta data descriptor though, in the end the actual stream transport will be something like RTSP, which is definitely supportes.

⁣Sent from TypeApp ​