jvcleave/ofxOMXPlayer #114: ofRPIVideoPlayer.h:5:25: fatal error: ofBaseTypes.h: No such file or directory

kr15h, 2018-01-10:
Just tried to compile with the openFrameworks master on Raspbian Stretch on a Raspberry Pi 3. Below a bit more error code.

```

In file included from /home/pi/openFrameworks-master/addons/ofxOMXPlayer/src/ofRPIVideoPlayer.cpp:9:0:
/home/pi/openFrameworks-master/addons/ofxOMXPlayer/src/ofRPIVideoPlayer.h:5:25: fatal error: ofBaseTypes.h: No such file or directory
 #include "ofBaseTypes.h"
                         ^
compilation terminated.
/home/pi/openFrameworks-master/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:358: recipe for target '/home/pi/openFrameworks-master/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofRPIVideoPlayer.o' failed
make[1]: *** [/home/pi/openFrameworks-master/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofRPIVideoPlayer.o] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks-master/addons/ofxPiMapper/example'
/home/pi/openFrameworks-master/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2

```

jvcleave, 2018-01-11:
what version of OF? Looks like it has been renamed to ofTypes.h in master

https://github.com/openframeworks/openFrameworks/tree/master/libs/openFrameworks/types

jvcleave, 2018-01-11:
or maybe ofGraphicsBaseTypes.h

https://github.com/openframeworks/openFrameworks/commit/2424116a06028c025b8e7145402d2b89be346c44#diff-b85a81715102dc0c0f6db53b074eb429

kr15h, 2018-01-11:
Will try those. Want to get all the master branches compiling. And also stable branches.

kr15h, 2018-01-11:
Probably depends on what types are you using there. I commented out the include and it went forward with the compilation.

kr15h, 2018-01-11:
`ofMain.h` is also included, could it mean that all the types used in the `ofRPIVideoPlayer.h` are referenced there?

jvcleave, 2018-01-11:
if `ofMain.h` works I would just use that. The include structure has changed in master so it probably wouldn't compile with previous versions of OF without the ofBaseTypes include.

kr15h, 2018-01-11:
This was adventurous. Too many things are changing. Next I will try the same old Jessie, with openFrameworks 0.9.8.