jvcleave/ofxOMXPlayer #116: Looking for minimal ofxOMX solution

kr15h, 2018-01-11:
I am looking for suggestions here. I would like to build a minimal ofxOMX player that would support only loading h246 movie with or without sound and play it back the same speed, be able to loop and render to ofTexture. I would appreciate hints on how should I get started with this. Maybe a link to the underlaying OMX header files and documentation would be it. Also, what steps of hardware configuration would it need.

kr15h, 2018-01-11:
I found an introductory description [here](https://elinux.org/Raspberry_Pi_VideoCore_APIs) and it looks like there are headers and example code in `/opt/vc`.

jvcleave, 2018-01-11:
you can look at `/opt/vc/src/hello_pi/hello_videocube` but it will only be able to play raw .h264 files (like the test.h264/big bunny video) on the pi

closed 2018-01-11 by jvcleave