jvcleave/ofxOMXPlayer #119: undefined reference to symbol 'eglCreateImageKHR'

PTS93, 2018-03-19:
I'm currently getting the following linker error:
`/usr/bin/ld: /home/pi/of_v20180319_linuxarmv6l_release/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayer.o: undefined reference to symbol 'eglCreateImageKHR'
//opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
`
I don't get much from this error. Any clue what could cause this?
Using Rasbpian Stretch with the latest OF armv6 nightly.

PTS93, 2018-03-19:
I had to apply the fix for https://github.com/jvcleave/ofxOMXPlayer/issues/115
and run sudo apt-get install libssl1.0-dev to even get there. Otherwise it would throw those issues mentioned before.

jvcleave, 2018-03-19:
I haven't tried Stretch yet but my guess is that maybe a library name has changed somewhere?