jvcleave/ofxOMXPlayer #120: Streaming Youtube Video

HowellPendragon, 2018-04-06:
Hey jvcleave,
i can't find out why i can't stream Youtube-Videos. I am extracting the Youtube-Video-Link with youtube-dl.
Other Hoster like streamcloud.me works fine.

e.g. "youtube-dl -o -f best https://www.youtube.com/watch?v=_LaVvGlkBDs" and i can play it without problem in omxPlayer.

but when i try to load it with your addon ofxOMXPlayer i get these Error : 
"FAST PATH MOVE OPEN FAILED - LIKELY A STREAM, TRYING SLOW PATH"
.....
"engine -> setup Fail" and just getting a white screen.

specs:
raspberryPi 3
memorySplit 512

Regards
Howell Jenkins

HowellPendragon, 2018-04-06:
Downloading the Video and playing it works fine.

So maybe there is something with session/cookies that youtube does not allow to stream them with only the link ?

Regards
Howell Jenkins

jvcleave, 2018-04-06:
yeah - youtube has broken that functionality a few times. youtube-dl is pretty good at chasing the fixes so not a bad way to work around it

HowellPendragon, 2018-04-06:
Thank you for the fast reply.
My question is, why does it work in the omxPlayer but not in your addon. I don't know what i should search for. Maybe omxPlayer got cookieAcceptance and your addon does not use cookies at all ?

So is this error just because of the cookies/session ? I will try it at least on tueseday.

Or do you have any other ideas ?

Best Regards
HowellJenkins

HowellPendragon, 2018-05-31:
Working well now with your latest update for openframeworks 0.10 and raspberry stretch.

good job and regards

closed 2018-05-31 by HowellPendragon