jvcleave/ofxOMXPlayer #125: Assembler errors when compiling example-basic

mysticcircuits, 2018-05-17:
Hello, hope all is well.

Been trying to suss out this library for a minute.  Apologies if this a noobish question, I am new to open frameworks and especially to OF addons so I am in over my head a bit when the compiler starts spitting out assembler errors.

Running Raspian Jessie on an RPI B+ v2, open frameworks 0.9.8 with the stable release of ofomxplayer for the same version installed as well.

When I try to compile I get the following messages after the normal compiler messages:

> {standard input}: Assembler messages:
> {standard input}: 15703: Warning: partial line at end of file ignored
> {standard input}: 15677: Error: undefined local label '.L4173'
> {standard input}: 15678: Error: undefined local label '.L4173'
> g++: internal compiler error: Killed (program cclplus)
> Please submit a full bug report
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
> ../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:3
> 36: recipe for target '../../../addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/BaseVideoDecoder.o' failed
> make[1]: *** [../../../addons/obj/linuxarmb6l/Release/ofxOMXPlayer/src/BaseVideoDecoder.o] Error 4
> make[1]: Leaving directory '/home/pi/openFrameworks/addons/ofxOMXPlayer/example-basic'
> ../../../libs/openFrameworksCompiled/project/makefilecommon/compile.project.mk:1
> 25: recipe for target 'Release' failed
> make: *** [Release] Error 2
> 

Not entirely sure what is causing this or what to do next.  If you need more info from me like the crash report or anything let me know.  Again I am a beginner with OF (although I have been programming in C for some time) so I am sure that I am just doing something dumb.

Thank you for your time.

Eli

jvcleave, 2018-05-17:
It sounds like the compiler is running out of memory. Try reducing the GPU memory (allocating more to the CPU)

jvcleave, 2018-05-18:
it's strange that it says `ofxOMXPlater` but otherwise I would try `make clean` first

mysticcircuits, 2018-05-18:
Yes sorry I am copying the error messages by hand to my other computer because the pi takes forever to load things on the internet.  That is why there might be typos

Make clean did the trick thank you.  Getting some other errors when running the example but I feel much better equipped to fix them now.