jvcleave/ofxOMXPlayer #127: Error when compiling example-direct-mode

CyrCom, 2018-06-06:
Hello,
I am clueless while I try to compile this example, help appreciated 👍 
```
/usr/bin/ld: warning: libssl.so.1.0.2, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libcurl.so, may conflict with libssl.so.1.1
/usr/bin/ld: ../../../addons/ofxOMXPlayer/libs/linuxarmv6l/ffmpeg/lib/libavformat.a(tls_openssl.o): undefined reference to symbol 'SSL_load_error_strings@@OPENSSL_1.0.2d'
//usr/lib/arm-linux-gnueabihf/libssl.so.1.0.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:403: recipe for target 'bin/example-direct-mode' failed
make[1]: *** [bin/example-direct-mode] Error 1
make[1]: Leaving directory '/home/pi/openFrameworks-20180402/addons/ofxOMXPlayer/example-direct-mode'
../../../libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:124: recipe for target 'Release' failed
make: *** [Release] Error 2
```

CyrCom, 2018-06-06:
Sorry, I see that Krišjānis had the same error here:
https://github.com/jvcleave/ofxOMXPlayer/issues/115
Is there a fix for it ?

Regards

jvcleave, 2018-06-06:
If you are using Stretch/OF 0.1.0 try using the master branch and running `sudo install_depends.sh`

I ran into this while updating it for Stretch/OF 0.1.0 and removing the custom/static ffmpeg solved the issue for me. It probably still needs some testing but was able to link using the libav packages from the Raspbian repos

CyrCom, 2018-06-06:
Thanks !! It Works :)

`sudo install_depends.sh` doesn't execute neither `sudo ./install_depends.sh` but I just copied and paste the inside.

I had already installed libavcodec-dev as suggested here, but not the rest.

closed 2018-06-08 by CyrCom