jvcleave/ofxOMXPlayer #134: draw rectangle without texture mode

micha1pu, 2018-10-10:
Is it possible to draw in direct video mode? Omxplayer can set video render layer number via --layer. When I run sketch with direct video mode I can see for 0.5s "bottom" layer with my rectangle, but rectangle is being overdraw by video. Texture mode is not fast for 1080p video.

jvcleave, 2018-10-10:
There is an [`omxPlayer.setLayer`] (https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayer.h#L431) but I haven't tested it much

Yeah - As mentioned in the Readme, texture mode works best at 720p

micha1pu, 2018-10-11:
I'm getting Segmentation fault when omxPlayer.setLayer(1) in setup. But anyway i don't think it will force video layer to bottom and OF will draw on top of it, they will be still on same layer i think.

jvcleave, 2018-10-11:
the layer property is probably something that should be passed via settings to make sure the player is ready to accept it. in the meantime you should be able to set it in draw() to avoid the exception

I was messing with it a bit and was able to change it but had to tweak the window settings in order to layer things properly. I'll post an example once I get some things worked out

jvcleave, 2018-10-12:
I added a layer option to `ofxOMXPlayerSettings` and [example-direct-mode-advanced](https://github.com/jvcleave/ofxOMXPlayer/blob/master/example-direct-mode-advanced/src/main.cpp)

I put everything in one file as it is important to see the window setup code in `main()`

This also requires a mod to OF so see [the note](https://github.com/jvcleave/ofxOMXPlayer/blob/master/example-direct-mode-advanced/src/main.cpp#L96)

closed 2018-10-12 by jvcleave

micha1pu, 2018-10-12:
It's working perfectly, thank you.

PTS93, 2018-10-12:
that's really great!