jvcleave/ofxOMXPlayer #135: I want to add const prefix some getter functions as possible

leico, 2018-10-23:
Hello jvcleave, thank you to create a awesome addon!

Now I trying to create apps with this addon.  When read this addons source codes, I got worried that getter functions returned non const values, also they are not const function.
I prefer a const modifier, it helps compiler optimization. ah, const becomes a slightly more complicated code.
Which is preferable ?

If you like, I will send PR like: 1425b0713e4fe0355368f360d9cbdd4184eaaf9b , also I will try add more const modifier as possible as.

Thank you for reading.

jvcleave, 2018-10-23:
Thanks but I am not a huge fan of const as I am often trying to work around it with other libraries.

As far as I have found it doesn't effect performance (the compiler does it's checks and deletes it).

leico, 2018-10-24:
Thank you quick reply.

It seems I had generated religious war unintentionally, I planned to use getters in other const functions with other library const functions...

It may not be a good method ( it becomes complicate ), how do you think to use overloads? like:

```cpp
      int some_getter ( void )       { return value; }
const int some_getter ( void ) const { return value; }
```