jvcleave/ofxOMXPlayer #137: more than 7  videos in data/sources/video folder cause segementation fault

zetteeh, 2018-11-28:
hey,
one one of the raspi 3+B models i use i've encountert a strange problem. 
no matter what i do. the 8th video in my list can never be played. it always crashes before and returns a segmentation fault error. 

dosn't matter which video is the 8th, there is always  the same result..
any ideas? 
thanks!

jvcleave, 2018-12-14:
Do you see this with `example-playlist`? With this example you can start it and press `n` to load the next movie. I don't have the B+ but I am not seeing the issue after loading 8 movies

gepatto, 2018-12-15:
@zetteeh Just tested this on a 3B+ with the example-playlist:
no problems here.
Are you using this example too?
If not, check for out of bounds like PTS93 said.

closed 2018-12-16 by jvcleave