jvcleave/ofxOMXPlayer #138: calling close() while playing in direct mode with audio will throw segmentation fault

gepatto, 2018-12-14:
Calling close() while playing a video in direct mode with audio will throw segmentation fault.
When you disable audio, `setings.enableAudio=false` it doesn't crash.

The same segfault happens when the video is done playing at the end and you call close()

Tested on 'example-direct-mode' with OF 0.10.1 on a rpi3b+ (256mb GPU split)

gepatto, 2018-12-15:
Oh I see I gave the wrong example name. I was using example-playback-controls, set `settings.enableTexture = false;` and while the video was playing pressed the 'q'  key.
It's playing the default BigBunny mov

It's openFrameworks  0.10.1 on a fresh latest raspbian, 
with the latest git commit of ofxOMXPlayer (but tried latest release as well)  
on a Pi3B+ with a 1280x800 (touch) monitor without internal speakers! 

I have another pi here, a Pi3B, still running of 0.10.0 
I'll do some more testing and try your code example above as well.

gepatto, 2018-12-15:
ok, using the example-direct-mode on openframeworks 0.10.1:
I have to set either `enableAudio` or `useHDMIForAudio` to false  (monitor without internal speakers!)
otherwise nothing is drawn to screen but it doesn't segfault when using your piece of code above.

jvcleave, 2018-12-15:
ok cool - what I am guess is happening is the audio player is failing to open because of the monitor type. It's not in the example but ofxOMXPlayer::setup returns a bool so you could check via

```
bool didOpen = omxPlayer.setup(settings);
```

gepatto, 2018-12-15:
Ah!
Now I see that if I use the same bool construct as `doPause`  
and call `omxPlayer.close();` in update () 
it doesn't segfault.

probably a Threading thing?

jvcleave, 2018-12-15:
yeah - often you don't want to do things on key press if there are graphics involved. it's safer to set a variable and then do the method in update or draw (so it happens on the main thread)

gepatto, 2018-12-15:
Ok thanks for the help!

The omxPlayer.close() on pressing 'q' is there in the example-playback-controls on git. 
I think we should it remove it there and use a `doClose` bool construct.
would you like me to Fork an PR?

closed 2018-12-15 by gepatto

jvcleave, 2018-12-16:
thanks - I may do it internally

e.g. set a `doClose` variable in `ofxOMXPlayer::close`

then actually close here:

https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayer.cpp#L313

However I am a bit worried how it would effect the destructor as I don't think it would wait for the onUpdate call. Will have to test