jvcleave/ofxOMXPlayer #139: audio low level or distorted

dasoe, 2019-01-14:
Hi jvcleave,

**thanks for this great addon I am heavily using for years now!**
I just upgraded a much used media player that w e developed for use in a museum. New version based on raspbian stretch, OF 0.10 and new (release version) of ofxOmxPlayer. All works very well - except Audio (HDMI) starts to be distorted when volume exceeds 0.5 %. I also did direct compare old an d new player: audio on new player is quite low when not distorted.
Might this be a problem in ofxOMXPlayer or did I oversee something?

thanks for any hints
oe

jvcleave, 2019-01-14:
not really sure - I currently don't have a way to check against the old

my guess would be to look around here:

https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofxOMXPlayerEngine.cpp#L1098

dasoe, 2019-01-14:
I see, thank you. I think leveling is not so much a problem. Rather the maximum possible volume I can get without disturbances seems quite low to me. A HDMI Screen with audio I used before is near to unusable now. But leaving old/new comparison aside: am I the only one out there with the problem "low volume"? Might as well be a global jessie->stretch problem, I just don't find anything...

thanks for any hints
oe

dasoe, 2019-01-24:
I think after some chekcing this problem is related to the core. Not sure if it would be a good idea to restrict the maximum volume in ofxOmxPlayer due to a problem outside, that might be temporary (so that we later can not use full power).
Also no one else reported. So closing the issue. Thanks!

closed 2019-01-24 by dasoe