jvcleave/ofxOMXPlayer #143: Jack output with `ofRPIVideoPlayer`

xseignard, 2019-03-11:
Hello,
Thanks for this addon!
I have a small question: how to use jack output with the `ofRPIVideoPlayer` ? Because of the this [line](https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofRPIVideoPlayer.cpp#L40), it forces the HDMI audio output, so for now I replace the value in your addon, but it's not a good long term solution. Any idea for a cleaner solution ?

Regards
Xavier

jvcleave, 2019-03-11:
You could put everything in your own settings object and use `openOMXPlayer`

https://github.com/jvcleave/ofxOMXPlayer/blob/master/src/ofRPIVideoPlayer.cpp#L53

xseignard, 2019-03-11:
Yes but doing so tends to break the goal of the `ofRPIVideoPlayer` being a (almost) drop-in replacement for the classic `ofVideoPlayer`.

Anyway, thanks for the tip!
Regards

closed 2019-03-11 by xseignard